ftCommunity/ftcommunity-TXT #1: TxtControlMain crashes when running RoboPro files from file system

rkunze, 2016-03-23:
When a previously downloaded RoboPro file is loaded and run in the TxtControlMain GUI, TxtControlMain crashes with a SIGILL (attempt to execute an illegal or privileged instruction).

The same RoboPro file runs fine when run in RoboPro online mode or downloaded and directly exceuted from RoboPro.

ski7777, 2016-03-28:
i have an idea:

When i go to FT-GUI/Settings/Language and i set it from DE to EN it changes but on next start ot FT-GUI it switches back. (same with master/EXT). But when i change one of these settings in normal firmware i can see them in our FW, too. Mybe FT-GUI has no write priveleges to the Flash.

ski7777, 2016-03-28:
i tested in fstab by changing /dv/ubi0_0 /rom set options from ro to rw.
Now settings will be keept but Flash-programms wont start like before.

rkunze, 2016-03-29:
The original firmware is mounted read-only by design, in order to minimize the risk of damaging anything. That settings are not accepted (and programs cannot be downloaded) is a known consequence of this, the planned solution is to bind-mount writable directories to the appropriate places where possible, and document the behaviour (i.e. settings are not permanent in the community firmware) where not possible.

However, as far as I can see this has nothing to do with the TxtControlMain crash when executing a stored RoboPro programm.

rkunze, 2016-03-29:
Ooops - accidentally hit "Close and comment" instead of "Comment"...

ski7777, 2016-03-29:
My backend-Idea was that RoboPro-Main maybe needs write access to the /rom/opt/knobloch/-Folder but it isnt the Problem.

rkunze, 2016-04-30:
This will definitely not be fixed before 0.9. May be fixed in 1.0 by implementing our own ROBOPro interface instead of using TxTControlMain from the original firmware.

harbaum, 2016-05-11:
Fixed by https://github.com/ftCommunity/ftcommunity-TXT/commit/a982f8fbc8ff79cead6430a3395809b939720198

closed 2016-05-11 by harbaum