ftCommunity/ftcommunity-TXT #3: Read basic config from FAT-Partition

TiniTech, 2016-03-25:
Make basic features configurable via config-file placed on FAT partition so the config can easily be edited and saved to survive firmware updates.

I'd particularly like to use this to set up WiFi so I can enter the WPA/WEP-Key on the desktop computer.

harbaum, 2016-04-17:
User installed apps are now stored on the sd card as a first step. This makes sure they aren't affected by updates of the root file system.

rkunze, 2016-04-24:
As of commit  6ab5545195adff72a6004918f1a57ad5135d6ba2, WLAN config for wlan0 is now stored on the SD card, in "config/net/wpa.wlan0.conf". Format for this file is the same as for wpa_supplicant.conf.  A basic configuration can simply set SSID and PSK for one network:
```
network={
	ssid="MY_SSID"
	psk="my secret wpa password"
}
```
If this file does not exist, an empty file is created automatically on startup. Config changes using the "WiFI" app are saved to "config/net/wpa.wlan0.conf" as well.

harbaum, 2016-04-24:
That's commit cecfd426

Sounds good, I'll give it a try asap.

rkunze, 2016-04-24:
> That's commit cecfd42

Yes, of cource - i accidentally pasted the parent commit :blush:

rkunze, 2016-04-30:
@TiniTech is the solution from cecfd4260258b70b8ed246fdbeefe62965a36bf9 OK for you? If so, can you close the issue?

TiniTech, 2016-05-01:
Sure, I think this approach separates data/config and system in a way that makes updating easy and saves (me and others) configuring the wifi over and over again.
I wasn't aware it was up to me to close the issue. Now I did.

rkunze, 2016-05-01:
I could have closed it myself, but because you opened the issue I thought it better if you give the final OK :smile:

TiniTech, 2016-05-01:
Thanks! 
However, to be honest I feel I was too quick on that...  Now that I understood the (new) principle with the users and the passwords for root, I have to deal with the fact that every system update overwrites the user/password configuration.
Would it be possible to also store config/etc/passwd and config/etc/shadow in the userspace on the FAT?

rkunze, 2016-05-02:
If you use the "simple layout", all changes to files in /etc/ will be stored on the FAT partition (in config/etc/).

TiniTech, 2016-05-02:
Would it be possible to implement that for the advanced layout as well? I like the idea of having (more of/"all" of) the config being separated from the system to preserve it during updates.

rkunze, 2016-05-03:
Yes, but doing so would negate some of the advantages of the advanced layout - namely, write speed and (more) robustness against unclean shutdown.

What I'm thinking about is a third layout variant, with uImage, am335x-kno_txt.dtb and rootfs.img and maybe some configuration defaults (like an initial WLAN config) on the FAT partition, and changes (downloaded apps, config changes, ...) on an EXT4 partition.

harbaum, 2016-05-03:
Is that really worth the effort? I am even not sure if we should release the files for the advanced layout. Having different files being part of different setups will probably confuse some users.

rkunze, 2016-05-03:
I think it is, especially since it would combine the advantages of the advanced layout (more robust, faster) with the advantages of the simple layout (config changes persist across upgrades). The disadvantage of this setup is that changes cannot easily be accessed "offline" from Windows, because these changes would be on an EXT4 file system.

ski7777, 2017-07-08:
Does anybody need this? Maybe we should just create a tool, which can transfer settings like WLAN via USB.

LarsK1, 2017-08-16:
Is that idea still necessary? (We intriduced a new layout, so do we need it anymore?)

PeterDHabermehl, 2018-05-15:
Obsolete due to new unified sd card layout

closed 2018-05-15 by PeterDHabermehl