ftCommunity/ftcommunity-TXT #4: Make system more end-user-friendly

TiniTech, 2016-03-26:
Right now all apps are equally accessible / deletable by the user ftc.

I would suggest the following:
* introducing an additional "end user" account (eg. ftc-user) which after logon via sFTP/SSH only "sees" the (user)apps directory (seebelow)
* storing apps in two separate locations: (1) for system apps (currently WiFi, NetInfo, FT-GUI, About) which are invisible to and undeletable by ftc-user and (2) for the app samples and apps generated by the end user.  
    Then a "cleanup" of the app directory doesn't remove basic functionality.

The idea is to allow beginners to gain experience slowly and not to confuse them with a full view of the linux fs, while still allowing the pros full access.

harbaum, 2016-03-26:
A tfc user already exists. Currently the whole gui runs under root permissions. The goal is to run it under the ftc user account. No need for separate directories then as the system apps could then be given root ownership and the ftc (or www) uder would not be able to delete them.

harbaum, 2016-04-04:
Done

harbaum, 2016-04-04:
We could  move ownership of the preinstalled apps from the ftc user to the root user to add some more protection. But since we are not providing a GUI to delete these apps there's imho no need for further protection.

closed 2016-04-04 by harbaum