ftCommunity/ftcommunity-TXT #5: Web-Interface improvements

TiniTech, 2016-03-26:
* allow categorizing the apps into subfolders via WebIF
* allow renaming of apps
* allow deletion of apps

harbaum, 2016-03-27:
Good ideas and definitely on the ToDo list. 

@ski7777  Please don't just check broken quick hacks.

ski7777, 2016-03-27:
@harbaum Now the broken feature is in extra repository.

harbaum, 2016-03-27:
Yes, it seems you created a local fork. Good. That's the way to go.

harbaum, 2016-04-01:
Deletion of apps has been implemented,

ski7777, 2017-03-01:
Maybe we could add a editor for Launcher V2 to edit the folder structure via web interface. Maybe we should show the apps in this structure.

Raphael

PeterDHabermehl, 2018-05-15:
Subfolders are also implemented, for my understanding, renaming Apps does not make much sense. So I would suggest to close this issue. Any thoughts?

PeterDHabermehl, 2018-05-21:
No thoughts - no issue.

closed 2018-05-21 by PeterDHabermehl