ftCommunity/ftcommunity-TXT #8: Windows is recognizing TXT as Serial Device and wont install normal TXT USB Driver

ski7777, 2016-03-26:
Everything is said

harbaum, 2016-03-26:
What kind of serial device? What driver is your PC using? Please be more specific.

ski7777, 2016-03-26:
it says "Serielles USB-Gerät (COM11)" and manufractor is "Microsoft"
[image]

ski7777, 2016-03-27:
Any ideas?

TiniTech, 2016-03-28:
I can confirm the following behaviour attaching the TXT via USB to my Win 10 Pro x64:

When the TXT is booted using the default firmware, it shows up as a Network adaptor, announcing USB VID 146A and PID 0015. 
*A quick search on http://www.linux-usb.org/usb.ids did not return any matches - according to the inf-File (under Windows x64 available  under C:\Program Files (x86)\ROBOPro\USB-Treiber Installation\TXTControllerontroller\Win7\fttxtusb_win7.inf) this VID is used by Knobloch GmbH.*

When the TXT is booted using the ftcommunity-TXT-FW a new device shows up which is listed as COM-Port. It has VID 0525 and PID A4A2. These IDs apparently belong to Netchip Technology, Inc. and their "Linux-USB Ethernet/RNDIS Gadget" (according to http://www.linux-usb.org/usb.ids).

harbaum, 2016-03-28:
Good point. We cannot use the vid/pid owned by knobloch. The vid/pid of the community firmware is probably the default one used by a stock linux kernel.

The "Linux-USB Ethernet/RNDIS Gadget" simply is name of this part of the linux kernel. We just need a windows10 driver for it.

harbaum, 2016-03-28:
You can either try to change the vid/pid in the original txt's inf file. Or you might try one like this:

http://www.yoods.co.jp/arm-y/linux.inf

TiniTech, 2016-03-28:
Changing VIDs in the inf-File makes Windows complain that the driver is unsigned - and unsigned drivers cannot be installed on Win 10 x64 by default. 

However, there may be a way to make windows install the right driver by itself:

A similar problem seems to have appeared for the community firmware for the Lego EV3; apparently the default MS RNDIS Driver included with Windows would work. (and actually, it appears that the Knobloch-Driver actually is a re-labelled MS RNDIS 6.0 Driver).

Maybe the solution found for the EV3 can be employed here as well - see Post by dlech, "Solution 2" here: http://answers.microsoft.com/en-us/windows/forum/windows_10-networking/windows-10-vs-remote-ndis-ethernet-usbgadget-not/cb30520a-753c-4219-b908-ad3d45590447.

The default RNDIS Driver apparently doesn't attach to particular VIDs but to MS-Specific "COMP" and "SUBCOMP"-Tags:

(snip from rndiscmp.inf taken from my Win 10 x64):
```
;when MSID is used for RNDIS Ethernet devices over USB transport
%RndisDevice%    = RNDIS.NT.6.0, USB\MS_COMP_RNDIS&MS_SUBCOMP_5162001
;when CompatID is used for RNDIS Ethernet devices over USB transport
%RndisDevice%    = RNDIS.NT.6.0, USB\Class_EF&SubClass_04&Prot_01
```

harbaum, 2016-03-28:
Lego users have the same problem with their ev3's:

http://www.lejos.org/forum/viewtopic.php?t=8020

This seems to be a windows problem and not a problem specific to the community firmware. Thus I am closing this bug. If you find a solution then please write a wiki page about it.

closed 2016-03-28 by harbaum

ski7777, 2016-03-28:
I hope we will find a solution but currently its working fine witj RoboPro by using set other IP option.

Raphael

P.S.: I cant write in the wiki because i havent write rights.

TiniTech, 2016-03-28:
Solved. Wiki in the making. Get driver from here & update COM-device with the driver: http://catalog.update.microsoft.com/v7/site/ScopedViewRedirect.aspx?updateid=37e35bd4-d788-4b83-9416-f78e439f90a2 
No further configuration required, IP is the default TXT-IP 192.168.7.2.

harbaum, 2016-03-28:
Excellent. Thanks a lot.

I'll still have a look at 'solution2' which would solve the issue without the need to install another driver if I understand this correctly.

ski7777, 2016-03-28:
jipi. It works fine on first try. Put it in the repository.

TiniTech, 2016-03-28:
If 'solution2' would lead to a plug-and-play experience under Windows 10 as well, this would be much appreciated.

I would however consider this an issue of minor relevance: I personally don't see the relevance of the USB connection, particularly as the TXT automatically signs on to the home WLAN once its configured. 

Things might look differently if the USB connection proved to provide a more reliable connection and/or a higher data rate than WiFi allowing eg. for trasfer of more or higher resolution webcam images for processing on a more powerful desktop computer.

ski7777, 2016-03-29:
In Win 7 the same problem appears. The TXT is directly recognized as RNDIS... but with no driver found. Installing the Driver from MS Catalog works fine and then the Computer gets his IP directly.