ftCommunity/ftcommunity-TXT #21: libfakeroot.so not found

ski7777, 2016-03-29:
```
fakeroot: preload library `libfakeroot.so' not found, aborting.
fs/tar/tar.mk:13: recipe for target '/ftcommunity-TXT/output/images/rootfs.tar' failed

```

I tried to build yor update but make gives me the following error. I didnt change anything on the VM.

Raphael

harbaum, 2016-03-30:
What exactly are you trying to do? Build the firmware? And what update do you refer to?

Did you try a full rebuild?

ski7777, 2016-03-30:
Ich schreib jetzt mal auf Deutsch:
Ich wollte euer Update mit dem App-Delete und co auf den TXT bringen. Also hab ich mit `git pull
make fischertechnik_TXT_defconfig
make`
den Buildvorgang angestoßen (die Maschine hatte vorher schon mehrmals ohne Problem die Firmware gebuildet) und dann kam die Fehlermeldung.

harbaum, 2016-03-30:
Try 
  make clean all

The buildroot manual is a good information source for problems with buildroot:
https://buildroot.org/manual.html

ski7777, 2016-03-30:
make clean all is running
Should i try with make fischer.... and make after that?

harbaum, 2016-03-30:
Afterwards the new setup has been built. No need for further commands.

harbaum, 2016-03-30:
Please close this bug if i's now woeking

ski7777, 2016-03-30:
currently buildroot is working. so wait

ski7777, 2016-03-30:
Problem solved by itself

closed 2016-03-30 by ski7777