ftCommunity/ftcommunity-TXT #22: After Adding/Deleting an App the Changes are only visible after pressing F5

ski7777, 2016-03-30:
Most is said:
After adding an App I come back to MainPage but without the new App. After pressing F5 it is visible.
After deleting an App I come back to MainPage but with html "No Image"-Placeholder and Name of the App on Positon. It goes away by pressing F5.

closed 2016-03-31 by harbaum

ski7777, 2016-03-31:
PROMBLEM NOT SOLVED in Chrome for Desktop!!!

harbaum, 2016-03-31:
Please stop shouting. That's impolite.

harbaum, 2016-03-31:
Can you please be more specific? Is the problem solved for other browsers? Or did you only test chrome? 

And are you sure the updated package has been rebuilt? Did you do a full rebuild?

ski7777, 2016-03-31:
The problem appears only in Google Chrome :-1:  But in edge and iexplorer it doesnt appear :+1:

harbaum, 2016-04-01:
Should now also work in Chrome.

ski7777, 2016-04-01:
Works Fine!!!
Problem Closed by my side.

commit 192f3612 (2016-03-31): Attempt to fix #22
commit 4ae8abdd (2016-03-31): Next attempt to fix #22