ftCommunity/ftcommunity-TXT #23: I2C Interface

chehr, 2016-04-06:
No I2C Communication with RoboPro.

harbaum, 2016-04-07:
The robopro user is member of the i2c group and should have the necessary permissions. Is your i2c device detected using the regular i2c tools like i2c-detect?

chehr, 2016-04-07:
I checked it with an Osziloscope which can decrypt the serial communication. But neither clock nor data is visible, only the static 3.3V.
I am not familar with a Linux command for I2C detection. Please advice.

harbaum, 2016-04-07:
A scan should show your device as well. The following is from a stock TXT with CE firmware.

```
# i2cdetect -l
i2c-0   i2c             OMAP I2C adapter                        I2C adapter
i2c-1   i2c             OMAP I2C adapter                        I2C adapter
```

```
# i2cdetect -r 0
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-0 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
# i2cdetect -r 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

harbaum, 2016-04-07:
Ah. Seems user ROBOPro is missing permissions:

```
# su ROBOPro
# i2cdetect -l
i2c-0   unknown         OMAP I2C adapter                        N/A
i2c-1   unknown         OMAP I2C adapter                        N/A
# i2cdetect -r 0
Error: Could not open file `/dev/i2c-0': Permission denied
Run as root?
```

It seems i2c device entries are not mapped to the i2c group. A udev rule should solve that. But not today ...

harbaum, 2016-04-08:
New udev rules seem to fix the permission problem. Please check if it also solves the problem with ROBOPro.

```
# su ROBOPro
# i2cdetect -l
i2c-0   i2c             OMAP I2C adapter                        I2C adapter
i2c-1   i2c             OMAP I2C adapter                        I2C adapter
```

chehr, 2016-04-09:
I would like to test the update, however currently I have the same issue ("setup.py", line 19) as Lars, means can´t built the newest update from GIT:
see following make message
`# make
/usr/bin/make -j1  HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" silentoldconfig
make[1]: Entering directory '/home/christian/ftcommunity-TXT'
BR2_DEFCONFIG='' KCONFIG_AUTOCONFIG=/home/christian/ftcommunity-TXT/output/build/buildroot-config/auto.conf KCONFIG_AUTOHEADER=/home/christian/ftcommunity-TXT/output/build/buildroot-config/autoconf.h KCONFIG_TRISTATE=/home/christian/ftcommunity-TXT/output/build/buildroot-config/tristate.config BR2_CONFIG=/home/christian/ftcommunity-TXT/.config BR2_EXTERNAL=support/dummy-external HOST_GCC_VERSION="5" SKIP_LEGACY= /home/christian/ftcommunity-TXT/output/build/buildroot-config/conf --silentoldconfig Config.in
make[1]: Leaving directory '/home/christian/ftcommunity-TXT'
>>> python-pillow c01f51dfff07416aba5f9bd78717be1887f67868 Building
(cd /home/christian/ftcommunity-TXT/output/build/python-pillow-c01f51dfff07416aba5f9bd78717be1887f67868//; PATH="/home/christian/ftcommunity-TXT/output/host/bin:/home/christian/ftcommunity-TXT/output/host/sbin:/home/christian/ftcommunity-TXT/output/host/usr/bin:/home/christian/ftcommunity-TXT/output/host/usr/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games" PYTHONPATH="/home/christian/ftcommunity-TXT/output/target/usr/lib/python3.4/sysconfigdata/:/home/christian/ftcommunity-TXT/output/target/usr/lib/python3.4/site-packages/" _python_sysroot=/home/christian/ftcommunity-TXT/output/host/usr/arm-buildroot-linux-gnueabihf/sysroot _python_prefix=/usr _python_exec_prefix=/usr  /home/christian/ftcommunity-TXT/output/host/usr/bin/python setup.py build  )
Traceback (most recent call last):
  File "setup.py", line 19, in <module>
    from setuptools import Extension, setup, find_packages
ImportError: No module named 'setuptools'
package/pkg-generic.mk:195: recipe for target '/home/christian/ftcommunity-TXT/output/build/python-pillow-c01f51dfff07416aba5f9bd78717be1887f67868/.stamp_built' failed
make: *** [/home/christian/ftcommunity-TXT/output/build/python-pillow-c01f51dfff07416aba5f9bd78717be1887f67868/.stamp_built] Error 1
`

harbaum, 2016-04-09:
Seems python-pillow needs a dependency on setuptools ... hmm

I can build everything just fine ...

harbaum, 2016-04-09:
Can you try to install python setuptools on your host?

chehr, 2016-04-09:
I just made make clean... If it is not Running i will install the Setup...

Christian Hehr

chehr, 2016-04-13:
Dear Till,
good work, now I2C is running with the "ftcommunity-TXT" and ROBO Pro. Issue solved.

Thanks chehr

closed 2016-04-13 by chehr