ftCommunity/ftcommunity-TXT #24: RoboPro <-> LCD

chehr, 2016-04-06:
Text or control elements from RoboPro not visible on TXT LCD.

harbaum, 2016-04-07:
Do they become visible if you start the original gui from the launcher beforehand?

chehr, 2016-04-07:
Yes.
Following scenarios:
After downloading a programm from RoboPro to TXT and with ftCommunity GUI, text or control elements are not visible on LCD.
After downloading a programm from RoboPro to TXT and with original FT-GUI (either by original TXT or back from ftCommunity) the text or control elements are visible on LCD.
To go back from FT-GUI seems not possible from TXT itself- only possible by WEB I/F.
It should be that RoboPro should clear the LCD in case of the need.

TiniTech, 2016-04-07:
I just tried to produce the error myself:

1) Running programs in Online-Mode never shows any display output. This seems to be a problem of the ft stock ROM (cf forum.ftcommunity.de/viewtopic.php?f=8&t=3204&p=21724#p21687) - the behavior is identical under the stock ROM and under FT-GUI running under ftc-TXT.

2) On ftc-TXT, after manually starting the FT-GUI app, the program could be run in in Download mode and showed its display output.

3) If the FT-GUI app was not started before the download, RoboPro still announced a successful download to the TXT, however, no display output was shown. However, the program itself seemed to be running - it switched lights on and off as programmed in RoboPro.

NB for chehr: Any app running on the ftc-TXT can be terminated with a short press of the blue button. (in addition to the x in python apps and the WebIF.

harbaum, 2016-04-07:
Thanks for testing. 

1) is not a issue limited to  the community edition (ce). So we can ignore that.
2) That's how it's meant to work
3) In that mode all output of the original firmware is redirected to nirvana since the ce owns the screen. That's by purpose as we don't know when Robopro wants to draw onto the screen.

So it's the way it's supposed to be. Thus i'll close this.

Ah, btw: I have never seen RoboPro painting on screen. Can you make a screenshot of this using the screenshot feature?

closed 2016-04-07 by harbaum

TiniTech, 2016-04-08:
For illustration purposes the output of the "Flipper"-Programm intended for use with TX ElectroPneuvac:
[image]
(The ???? only show up because I don't have the expected inputs attached.)

And the drawing in RoboPro to generate the output:
[image]

harbaum, 2016-04-09:
Cool. Thanks.