ftCommunity/ftcommunity-TXT #25: "Missing" corners on Startup / Shutdown screen

TiniTech, 2016-04-07:
When booting up, the start screen shows a small black box in the top left corner. No big deal, but it hurts my eye ;)

When shutting down via long button press while the FT-GUI is running, the Shutdown-Screen leaves a bigger black field in the top right corner - showing the time the FT-GUI displays in this corner. Also no big deal but ... ;)

harbaum, 2016-04-07:
That's a known problem. The black square on the boot screen is actually the remains of some text the kernel tries to write to the console. If you connect a keyboard and type you'll see some more text during shutdown.

The fact that most of the time the "community edition" text is not being displayed on the splash screen seems also related. I'll one day have a look at that. Perhaps ...

And finally the black box in the top right border when shutting down with the original gui running may be part of the clock. I don't think i'll care for that. The whole old GUI will hopefully be gone one day.

harbaum, 2016-04-11:
Fixed the startup screen. The shutdown from within the oroiginal gui imho  cannot be fixed

closed 2016-04-11 by harbaum