ftCommunity/ftcommunity-TXT #28: Issues with Simple Layout SD-Card

TiniTech, 2016-04-27:
Re: readme.md: The name of the image-file should be output/images/rootfs.img (not: .image as it says in the text)
I'd also suggest to add a note that the first startup with the "Simple Layout" might take quite some time - at least for me it did. Subsequent startups felt a lot faster.

Re: Use: After startup the launcher hangs. I get to see the full launcher with all icons as I'm used to, but it doesn't react to any touches on the display.
The Web-Interface is however available, as is the SSH-Login. On the Web-IF it says "Launcher not responding".

When I install the firmware from the same build-process in the classic setup on my partitioned SD, everything works fine.

rkunze, 2016-04-28:
Re: startup time at first startup: At first run, the SSH daemon calculates a set of host keys - this takes a while (~20-30 seconds). On subsequent startups, the pre-existing keys (from /etc/ssh in the advanced layout or from /config/etc/ssh in the simple layout) are reused. But this is independent of the SD card layout, i.e. you should have this delay on the 2-partition layout as well.

TiniTech, 2016-04-30:
It appears that both, the boot time and the hanging were issues with the SD-Card I used. A full re-format fixed both issues. Now even key creation didn't really cause any noticeable holdup anymore.

And regarding the adaption of the Readme: I just found out that the file is editable online, so I fixed image->img myself.

I therefore consider the issues listed above solved.

closed 2016-04-30 by TiniTech