ftCommunity/ftcommunity-TXT #31: Add status icons (at least for network)

TiniTech, 2016-04-30:
It's rather onerous to always check with the NetInfo-App whether errors occuring are due to a missing network connection.

Suggested solution would be to also give the launcher a sandwich menu so the right part of the header could be used for status icons.
In the meantime the NetInfo app could be changed to display wlan0-info when launched (instead of the rather irrelevant Io-device) so at least the info on a potentially broken wlan0-configuration is visible immediately.

harbaum, 2016-04-30:
wlan0 is now being displayed by default in the netinfo app.

Adding a menu to the launcher ... hmm ... i don't know. But imho any menu added to the launcher should be on the left as usual for consistency.

TiniTech, 2016-04-30:
The menu certainly should be on the left. I intended to suggest:
- a sandwich menu on the left, opening the list of categories
- status icons on the right of the (then free) top bar of the launcher

ski7777, 2016-04-30:
I  think wie should add some buttons wäre the close button is located. These buttons should be '?' for help oft the APP. This should be a simple HTML viever to viev the app's HTML pages. The second button is to viev basic Information e.g. connectivity switch and indocator, USB connected indocator...

TiniTech, 2016-05-01:
I agree on having status icons for Wifi/BT/USB links.
I strongly disagree on any sort of "help" function on the TXT-touchscreen itself:
- first of all, I think apps should be self-explanatory
- if they are not, they will need more explanation than I would want to read on the TXT, for everything else there is the Web-frontend of the TXT
- from experience: writing documentation is quite annoying to most developers: The cool part is to get things to work, therefore barely anyone will write a help text anyways, not to mention implement a context-sensitive help function
- noone expects such system-wide help function for embedded devices: neither iOS, Android nor WindowsPhone have such icons
- if any developer wants to implement a help function in any yet-to-be-developed app, they are free to include a help-entry in the hamburger menu on the top left corner

rkunze, 2016-09-25:
Release 0.9.2 incudes a status bar on the main launcher window with a network status icon.

closed 2016-09-25 by rkunze