ftCommunity/ftcommunity-TXT #32: App store: Improve error handling (just a little bit at least)

TiniTech, 2016-04-30:
Running the app store without network access leads to "Unknown Error: 3".

As "no network connection" seems to be a problem that could be common, it would be great to have a descriptive error message in this case.

TiniTech, 2016-04-30:
Turbo-tester TiniTech is still unhappy ;)

Two situations: no network, I open the app store: Error message is "Network error: Host not found". OK, great.
However - it takes quite long, until anything happens and I get to see a blank window for quite a long while, making it appear as if the store crashed.
-> I'd suggest displaying the "waiting indicator" also shown when an app launches (cool new design by the way, like it much more than the previous one)

Second situation: The app store is still open when the connection breaks. I then try to install an app:
Now I get to see an "unknown network error" - which is again not very specific. (OK, if the error code is not specific enough to say e.g. "Connection to the app store could not be established. Check your network link" then so be it.)

But: The error window has an additional Qt-Titlebar with the Qt-Logo on the left and an ? X on the right. THis definitely is not intended.

harbaum, 2016-05-01:
> But: The error window has an additional Qt-Titlebar with the Qt-Logo on the left and an ? X on the right. THis definitely is not intended.

This is supposed to be fixed, now.

> Now I get to see an "unknown network error"
You should now get an additional numerical error code. Please let me know which it is and i can add a more speaking text.

TiniTech, 2016-05-01:
Now I get the expected "host not found" error in both situations. However, if the link breaks after launch of the app store, it takes long (my feeling is: longer than before) until the error appears. In the meantime I could interact with the app store, eg. klicking install several times before the error occurred. 
I'd strongly suggest displaying a waiting indicator here to confirm that the press on install was actually recognized (for me this is not always the case, it seems my touch screen has issues in the corner every once in a while...)

harbaum, 2016-05-01:
Latest commit adds the busy indicator.

I wonder why you have a need for this. In my setup the busy animation only shows up for a fraction of a second and is barely visible.

TiniTech, 2016-05-01:
Thanks! If all is well I won't see the busy animation either.
But if things go wrong and there is no network connection, the app makes the impression to hang / not do anything before after a long wait the error message appears.

TiniTech, 2016-05-01:
OK, the waiting indicator now nicely bridges the time until the error message appears. 

This time in situation 2 (after breaking the connection by switching wifi off when the app store was open already) I get an "Unknown network error (99)" (I think before that there was the "host not found" which I would have expected again now.)

harbaum, 2016-05-01:
I didn't translate error 99 so far. But a quick look at the qt network error code table at e.g. http://doc.qt.io/qt-4.8/qnetworkreply.html#NetworkError-enum results in: "UnknownNetworkError"  :-)

So in this case my translation of the error code seems to be perfect!

commit 26a9a647 (2016-04-30): Display detailed network error messages. Fixes #32
commit 860a63c1 (2016-04-30): Display detailed network error messages. Fixes #32
closed 2016-05-01 by harbaum