ftCommunity/ftcommunity-TXT #34: NetInfo-App doesn'

TiniTech, 2016-04-30:
- structure

rkunze, 2016-04-30:
For USB, this is expected. The usb0 network interface is always brought up, and it is always configured to have IP 192.168.7.2, even if USB is not connected.

harbaum, 2016-05-01:
The netinfo App reflects the systems network config. It shows the IP settings of all interfaces which are currently up. That doesn't necessarily mean that the link is connected.

TiniTech, 2016-05-01:
Ah, I wasn't aware of that. I'd then suggest adding another line for the "Link status" in the app (and possibly an icon in the title bar (at least in the launcher, maybe even permanent in the title bar for all apps)

rkunze, 2016-09-25:
Release 0.9.2 incudes a status bar on the main launcher window with a network status icon.

closed 2016-09-25 by rkunze