ftCommunity/ftcommunity-TXT #37: Webscreenshit wohnt show the current screen

ski7777, 2016-05-01:


ski7777, 2016-05-01:
Oh,
i inly can See the Boot up screen in the Webbrowser

Raphael

TiniTech, 2016-05-01:
Kannst du bitte a) diese unsäglichen Typos korrigieren und b) erklären _welches_ Problem genau Auftritt und _unter welchen_ Umständen?

TiniTech, 2016-05-01:
re: initial issue: I can confirm the following behaviour using FireFox 46: When I click the "take screenshot" link on any app-page, the subpage opened shows the TXT boot-up screen. This is indeed irritating as I'd either expect no picture or a picture of the current screen.
However, when clicking the "Take screenshot" button, the displayed image is updated an shows the current screen as I see it on the TXT.

re: further improvements: Move the Screenshot Link (also) to the main page of the TXT-Web-IF as taking a screenshot is no specific feature of any app.

ski7777, 2016-05-01:
[image]
In Chrome on android and on Windows it looks like this for ever.

TiniTech, 2016-05-01:
Just noticed that the screenshot is updated automatically when I enable JavaScript for the Web-IF (hree my NoScript interfered...) Therefore seems to be a Client-Side problem with the JavaScript of the WebIF.

ski7777, 2016-05-01:
It wont work in all Browsers (Chrome, Edge, FireFox).

TiniTech, 2016-05-01:
Can't confirm from my side - running the advanced layout on the SD, tried FireFox with the correct NoScript settings, iPad default Browser and Chrome on Android. All automatically update the screenshot to show the (correct) current state of the TXT-display; changes on the TXT are displayed after the "Take screenshot"-button is pressed.

harbaum, 2016-05-02:
Cannot reproduce

closed 2016-05-03 by rkunze

harbaum, 2016-05-03:
What happened to the "default screenshot"? The purpose of it was to avoid an "broken image" icon to be displayed by the browser while no screenshot had been taken.

It's the same image still present at board/fischertechnik/TXT/rootfs/etc/ftc-logo.png Maybe one can copy that to /tmp  at boot time.

Also is /tmp a good place? Would the ftc users home make more sense as this would be persistent and could be extended into a full screenshot gallery one day?

rkunze, 2016-05-03:
Re "default screenshot": I was going to answer that the screenshot will be regenerated on first access, but then remembered that not everyone has JavaScript activated :smirk: 

Sure, we could copy a default screentshot to /tmp. But the screenshot function depends on JavaScript anyhow. Without JavaScript activated, no screenshots will be generated, and the only "screenshot" that will ever be displayed is the boot screen (with the old version) or a "broken image" (as of commit 776339fb9158098396ac5ce787bdaf46f237a965). So maybe a better approach would be to include an appropriate <noscript> section in screenshot.html that tells people to activate JavaScript.

Re placement in /tmp: As I see it, the typical usecase for the screenshot feature is "take screenshot, save screenshot image from browser", and permanent access to the image on the TXT is not really necessary. But if the rest of the community disagrees or if we extend it into a full gallery in the future, we can of course adapt the storage location.

harbaum, 2016-05-03:
Hi. No problem. I am fine with that. 

In the long term i'd like to see some generic setup with a place where files can be stored for rw access. This would then just be rw in the "advanced setup". In the "simple setup" this would consist of default files being placed in the ro file system but which can be modified through the rw overlay. It was my impression that the ftc users home works exactly like that, doesn't it? I must admit i haven't looked at the details of the simple setup at all ...

rkunze, 2016-05-03:
No, /home/ftc is just a bind-mount of /medi/sdcard/data in the simple setup. But /etc is a writeable overlay. But we're getting a bit off-topic here.

harbaum, 2016-05-04:
I added a link to the screenshot in /etc and some small javascript to use that as long as there's no screenshot in tmp.

rkunze, 2016-05-04:
I'm not sure that this will really work. See inline comment at ffd4055518936fe49739b77bd7a05c9389e3a9ce

ski7777, 2016-05-26:
Same Problem again. I can only see the default Screenshot: "Bootscreen"

commit 776339fb (2016-05-03): Write screenshots to /tmp/screenshot.png instead of /var/www/screenshot.png

Fixes #37

/var/www is now owned by root instead of ftc because changing the owner caused the image build to fail because the new symlink /var/www/screenshot.png -> /tmp/screenshot.png does not have a target in the build system, and the only reason why it had to be owned by ftc was writing screenshots.