ftCommunity/ftcommunity-TXT #38: Issues with Text-mode apps

TiniTech, 2016-05-01:
As ski7777 pointed out in the forum, text-mode apps cannot be terminated via home-button - one has to touch the X in the corner to close the app. Pressing the blue-lit home button has no effect on the running text-mode app. 

Also, a running text-mode app is not shown in the WebIF as running and cannot be terminated via the web-IF

If another app is launched via the WebIF, while the text-mode app is running, the text-mode app shows up again after the other app is terminated via WebIF.

harbaum, 2016-05-02:
Most of this has been fixed. Remaining issues: The app is considered not to run anymore if it has stopped but if the apps text output window hasn't been closed, yet. Also the window cannot be closed using the power button.

Both can be fixed but is not important as long as no apps are using this mode.

ski7777, 2016-12-12:
Maybe we could put the text mode system into a external framework. There we would have no problems with the button.

Raphael

harbaum, 2016-12-12:
I'd like to hear other opinions on that. If someone else also would lime to see this moved into a seperate app then I'd simply remove text mode support. It's basically unused and will probably not be missed.

harbaum, 2016-12-17:
Ok, i'll then remove text app support and Raphael creates an external framework for this.

harbaum, 2016-12-17:
I have removed all text mode related code. It's now up to ski7777 to come up with his new and improved solution.

closed 2016-12-17 by harbaum

ski7777, 2016-12-17:
👍 Now in my winter holidays this will be one of my projects.

Raphael

harbaum, 2016-12-17:
Looking forward to your new solution. Please don't forget to finish it before submitting pull requests and make sure it's documented.

ski7777, 2016-12-17:
I will commit it to a new branch of my fork. Comments will not be a problem.
I will just grab your old subprocess and text viewer handling and put it into a touchapllication.

Raphael

harbaum, 2016-12-17:
The idea is that a text mode app does not contain _any_ qt related code. That's why it's a called a text mode app. Please come up with some clever new idea. Just copy'n pasting the old code into each and every text mode app is nonsense.

ski7777, 2016-12-17:
I think this was a missunderstanding. I will put the qt code into the launcher. I will just replace your TextmodeDialogs TouchWindow to a TouchApplication.

Raphael

harbaum, 2016-12-18:
You talked about an _external_framework_ which definitely makes sense given the fact that up to now no single  text mode app has been written.

But this also means that it isn't useful to re-integrate the functionality into the launcher. And just replicating the old setup also isn't useful as we could then just have left everything the way it was.

ski7777, 2016-12-18:
I didn´t start yet with writing the framework, so I won´t start it. So I will delete the hello App from the store now.

Raphael

harbaum, 2016-12-18:
The "hello" app has been removed a few days ago ...

To summarize: You asked for the current solution to be replaced by an external framework. Then i removed the current solution in favour of your new framework. Now you don't like your own idea of an external framework anymore and text mode support is simply gone.

ski7777, 2016-12-18:
I am not sure whether anybody will use text mode apps. I will try to implement it.

Raphael

harbaum, 2016-12-18:
That's up to you. And it may indeed not be worth the effort. But your complaints about minor issues with the existing solution made me assume that you had a problem with it and that you had a need for an improved solution.

commit ede07608 (2016-12-17): Remove text mode support, fix #38