ftCommunity/ftcommunity-TXT #39: Sub-windows & behaviour of home-button (particularly: "About"-App)

TiniTech, 2016-05-01:
When the Version/Licenses sub-window is opened in the "About"-app, it shows an X in the top right corner. Tapping on the X only closes the sub-window and returns to the "About"-app.
Pressing the "Home"-Button in the same situation shows the launcher.

Is this intended? 

I'm uncertain myself right now - it irritated me because I expected the "home"-button to be a "tap-the-x"-replacement. On the other hand I think it's a good idea from a usability perspective to have the button always take one back to a known starting point - the Launcher.

harbaum, 2016-05-02:
Yes, this is how it's meant to be. The (X) closes a window. The power button closes the entire application. It's sort of an "emergency exit"

TiniTech, 2016-05-02:
Sounds good to me. I think this is better explained in the wiki, so I added a page for the "UI concept" and will close this issue.

closed 2016-05-02 by TiniTech