ftCommunity/ftcommunity-TXT #48: root priveliges for apps

ski7777, 2016-08-01:
The update script was finished a few weeks ago by me, but it only worked by using a shell with root. But currently the users have to put the SD-Card in the compute or use the root-access to update the operating system. I don´t think that this is the best solution, we have to search another Solution. In my opinion we should run the Launcher with root rights. Normal apps will be started with "ftc"-rights and apps with `root: True` in the manifest will only be started after the user have acknowledged, that the TXT could be destroyed.
What do you think?

Raphael

harbaum, 2016-08-01:
Why do you want to run the launcher with root permissions?

ski7777, 2016-08-01:
To run a app with root permissions.

harbaum, 2016-08-01:
The laucher does not need to have root priviledges for that. It can e.g. use sudo for that.

Does your update script have a full blown user interface? If not there's no useful way to run it from the launcher. And if you need to run it via ssh anyway then you can manually use sudo to give it root permissions.

So no, i don't see a reason to give normal user apps root permissions. But as always you are free to implement it, anyway.

ski7777, 2016-08-01:
Maybe we could make the settings APP able to set both passwords. And in the Update App we could a keyboard to enter the ftc or root password

Raphael

harbaum, 2016-08-01:
Why not? Just do it.

rkunze, 2016-09-25:
As of v 0.9.2, sudo is enabled by default for the ftc user (and by extension for all apps, since apps run as "ftc"). If no password has been set for ftc, sudo will request permission with a touchscreen dialog. If the ftc usre has a password, it is the gui apps responsibility to request the password and pass it to sudo.

closed 2016-09-25 by rkunze