ftCommunity/ftcommunity-TXT #52: Sound Issue

ski7777, 2016-09-23:
When trying to play Sounds via ftrobopy, this is only possible, if the ft-GUI open. If ft-GUI is closed, maybe the first sound will be played and then you can listen to the silence.

Raphael

harbaum, 2016-09-24:
Since when does that happen? Since the update of ftrobopy? Which application do you see the problem with?

harbaum, 2016-10-18:
No answer ...

closed 2016-10-18 by harbaum

ski7777, 2016-10-18:
Sorry for my late answer.  in the days after Convention I had lots of work and currently I am in holidays. Basicly some time after accessing  sounds the first time I can't access more sounds.
I don't know whether this problem appeared in 0.9 but in 0.9.1 and 0.9.2 i cant use sounds via ftrobopy. But I think we should use our new sound feature now. Maybe we could write a python module to access custom sounds easily.

Raphael