ftCommunity/ftcommunity-TXT #54: Store has trouble with "large" apps

TiniTech, 2016-10-16:
When installing the MP3-Player app, the waiting indicator gets slower and slower until it eventually stops moving at all, giving the impression that the system hangs. However, this is in fact not the case: Just waiting finishes the installation of the MP3-Player and returns to the app list of the store as expected.

Pushing the main TXT-button while the waiting indicator "hangs" terminates the store - but leaves the waiting indicator on the screen...

Fiddling around I even managed to corrupt the list of installed apps somehow (none of the previously installed apps showed up any more and any action in the store gave me a "could not delete app" or "could not create directory" error (to fix it, I deleted the whole app-directory from the SD-card.

I'm using the advanced setup of the SD-card and the 0.9.2-1-g8361410 firmware I built from scratch today.

harbaum, 2016-10-17:
The installation simply takes that long. There's not much we can do about that. 

The fact that the busy indicator slows down shows that the TXT is struggling quite heavily with this. I assume it's the SD card writing that slows the whole box down.

I also like to avoid doing a lot of stuff in the background. The TXT is just too weak for that. So the whole installation needs to be done by the store and not e.g. by some background process like on a modern smart phone.

There are a few things we could try:
- add some installation progress percentage display to help the user to be a little more patient
- Prevent the user from leaving the store app while the installation is in progress
- Don't distribute packages > ~1-2MB

What do you suggest?

harbaum, 2016-10-17:
Hmm ... it's the download itself that's so slow. Maybe I need to re-think the memory handling there ...

harbaum, 2016-10-17:
I've done some first optimizations. You can test them by just copying the updated store.py to your existing setup without having to recompile anything.

It's still missing the locking of the hole GUI during installation. But download should be faster now (not slow down) and there's some progress indicator. The first 50% are the download itself and the second 50% is the unzipping of the files. Please give it a try ...

harbaum, 2016-10-18:
Now the GUI is disabled during download. This will still require you to wait some time if the download and long and slow. But it's imho the only useful way to deal with this. Please give it a try.

TiniTech, 2016-10-18:
I really like the blurry design while the UI is disabled!

The download itself feels speedy (i.e. up to 50% everything goes smoothly), but then the extraction feels slow again - in case of the MP3-Player, reaching 52, 54, 56 and 58% takes long and does not show any movement of the busy indicator; it is only updated when a new extraction step is completed. 

Would it be possible to detach extraction progress and redrawing of the busy indicator, e.g. running a second thread which just (re)draws the busy indicator until told to stop by the installer? Or is it just the TXT being overloaded with copying/extracting files?

If it's only the TXT I'd suggest to add some more explanatory text, i.e "Downloading..." for the first 50% and "Installing.... This may take a while and make the TXT appear to hang. Please be patient."

I also thought about adding some "heavy=true" option to the manifest to explicitly display a corresponding warning message. 

Furthermore, I noticed that a push of the button (still) terminates the store-app - which may lead to corrupted installations. I'd suggest to disable this as well - particularly as a "real" hard reset remains possible by unplugging or by a very long button press should it ever be needed.

harbaum, 2016-10-31:
I don't think it's worth the effort spending a few hours to make the decompression a background process just to have the busy animation running faster.

harbaum, 2016-10-31:
Actually whatever you do should not corrupt anything. Installation of apps isn't more than just extracting a zip archive. This should not affect any apps already installed nor should an incompletely extracted archive cause any problems. Can you try to figure out what exactly goes wrong in that case?