ftCommunity/ftcommunity-TXT #55: USB keyboard doesn't work if GUI is run under FTC user

harbaum, 2016-10-31:
The USB keyboard works fine if the launcher and subsequent apps are run with root permissions. When being run under permissions of the ftc user only some keys do anything at all and even those generate wrong key events. This can best be seen in the "TXT" tab of the gallery app.

harbaum, 2016-11-06:
Things that don't help:

- invoking su with option -l to get a "login like setup"
- doing "chown ftc ``tty``" before starting the launcher to make sure the tty can be accessed by user ftc

ski7777, 2017-03-01:
Currently when typing something on a keyboard the character are shown in the background. This destroys the whole UI appearance. Screenshot will follow.

Raphael

ski7777, 2017-03-02:
[image]

harbaum, 2017-03-02:
Are you sure you didn't cause that yourself with the changes you did to rcS when enabling VNC?

ski7777, 2017-03-02:
This only occur with running the VNC Server, but without running the VNC Server only crypto characters are shown in QLineedits etc.

Raphael

PeterDHabermehl, 2018-05-15:
Is this still an issue?

harbaum, 2018-05-15:
Technically yes. But it seens nobody else cares and to me it's also more an annoyence then a problem. I'd consider this a 'close, won't fix'

harbaum, 2018-05-15:
Won't fix

closed 2018-05-15 by harbaum