ftCommunity/ftcommunity-TXT #59: Launcher: Reorganize app categories

rkunze, 2016-12-09:
@harbaum suggested in [a comment to PR#58](https://github.com/ftCommunity/ftcommunity-TXT/pull/58#issuecomment-266105256) to change the app categories for the launcher to:

* system (also used for the settings)
* models (for everything controlling models)
* tools (for things like clocks, calculators, ...)
* test (For demo and test stuff)

This issue is intended to discuss and hash out these changes.

rkunze, 2016-12-09:
Categories "system", "models" and "tools" look good to me. I don't really like "test", but the only name I can come up with for this category would be "demo", and I don't really like that either.

As for IDEs, the current plan for RoboBlocks and IPython/Jupyter is to not have them show up in the launcher at all, but trigger them via the web interface and have them run independently from the launcher. Brickly is an oddball there, because it's an app with builtin IDE, but I'd simply put in into "models", because controlling models is its main function.

I don't think we need an extra category for the "alien stuff" - existing "aliens" like the WeDo demo can just go into test/demo, other apps that use special stuff go into their respective categories, and popular support libs just go into the main image for now (just like ftrobopy). 

We can worry about a package system for common support libs when we have so many of them that putting them into the main image is no longer a viable option.

rkunze, 2016-12-09:
Another thought: The "all" category that is displayed on startup gets crowded pretty quickly. How about displaying the "models" category on bootup instead of "all"? Or make the initial category configurable?

harbaum, 2016-12-10:
I can simply make the current category selection persistent by storing it in a file. 

And yes, in the long term some icon based folder concept would be nice. The category in the manifest would then still be useful as a default folder. But the whole setup with creating renaming and deleting folders, moving apps between folders and automatically creating folder icons is pretty complex. That's for a later version.

There's another thing I missed: Translation of the app name and description. I have added a [de] section to the about apps manifest as an example. Any better ideas?

rkunze, 2016-12-10:
> I can simply make the current category selection persistent by storing it in a file.

Sounds good.

> Translation of the app name and description. I have added a [de] section to the about apps manifest as an example. Any better ideas?

Not really. Even though I'm not excited about adding yet another variant to the ever growing bunch of i18n variants we have around (PyQT, Blockly, whatever we're going to end up with for the parts of the web interface that aren't based on blockly, ...). But I think doing i18n right is a topic for later.

harbaum, 2016-12-12:
We'll end up with the pyqt mechanism, the manifest stuff and something for the web based things. Not too bad imho ... 

I am open to any better idea. But for a toy i think it's quite important to have a complete set of translations. I e.g. recently noticed that the sound names in the test section of the original firmware are english and my kids don't understand their names ...

harbaum, 2016-12-12:
I've implemented the default categories. This isn't fixed and can easily be adjusted should we come up with a better set.

I did some "auto adjustments" and e.g. apps referring to categrory "settings" will show up under system. Likewise apps requesting the "test" or "tests" category will show up under demos.

ski7777, 2016-12-12:
I think mixing  up a mechanism for pyqt and a mechanism for the python based web stuff is not a good idea. My system which uses a manifest like ini file is useable everywhere. I could add a locale support to get the locale from /etc/local. The, I think, killer feature is that you can edit the files with every text editor without needing extra tools.

Raphael

harbaum, 2016-12-12:
The ability to work with a text editor is indeed nice. But that's where the advantages end. Even using the text editor is a pain for any non-ascii languages. Your system simply isn't flexible. It e.g. doesn't even support all german umlauts let alone other languages. You'd have to update your system for every new language you want to support. Manually htmlizing everything because you want to avoid utf8 also doesn't make too much sense.

Qlinguist and the related tools on the other hand have been developed for a long time and are known to work for many languages in all kinds of environments. And yes, blockly has a different method of i18n. But it already comes with a bunch of complete translations. It'd be stupid not to use them.

harbaum, 2017-03-03:
The category system is gone ...

closed 2017-03-03 by harbaum