ftCommunity/ftcommunity-TXT #61: RTC Wake Alarm is not usable from out of an app

ski7777, 2016-12-09:
Currently `/sys/class/rtc/rtc1/wakealarm` is not writable for the ftc user.
For some applications it would be great to wake up the TXT periodically. I think we have two ways:

- Everybody can set his wakealrm and destroys the wakewalrm of another person. With this we doesn´t know which app to start after boot
- We have a app autostart handler, on which apps can register and unregister autostarts. If system is running it will start the apps the normal way. If it recognises that the TXT was bootet by RTC it will run all missed autostarts. But what to do if another (maybe autostarted) app is running? Maybe we could send also arguments via launches launch command, so the app could recognise, that it was autostarted.

Raphael

ski7777, 2017-03-01:
Anybody suggestions?

PeterDHabermehl, 2018-05-15:
No discussion within more than one year - obviously not required.

closed 2018-05-15 by PeterDHabermehl