ftCommunity/ftcommunity-TXT #62: Manifest parsing for apps breaks if the manifest contains umlauts

rkunze, 2016-12-10:
Commit 2a1713e910e79d628ec6dc81377366e6ce4e7cbf introduced UTF-8 umlauts ("Ü", to be specific) in the manifest of the "about" app. This causes mainfest.read() in launcher.py to fail with

```
Traceback (most recent call last):
  File "/opt/ftc/launcher.py", line 968, in <module>
    FtcGuiApplication(sys.argv)
  File "/opt/ftc/launcher.py", line 705, in __init__
    self.addWidgets()
  File "/opt/ftc/launcher.py", line 950, in addWidgets
    self.apps = self.scan_app_dirs()
  File "/opt/ftc/launcher.py", line 891, in scan_app_dirs
    manifest.read(manifestfile)
  File "/usr/lib/python3.5/configparser.py", line 696, in read
  File "/usr/lib/python3.5/configparser.py", line 1012, in _read
  File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 283: ordinal not in range(128)
```

(after deleting the try...except  around file loading in launcher.py)

harbaum, 2016-12-10:
I wonder why I didn't trigger that problem. Should be easy to fix anyway ....

ski7777, 2016-12-10:
I didn´t found a way to solve it, so I used the html umlaut tags and a simple converter

closed 2016-12-10 by harbaum

ski7777, 2016-12-12:
Why to close a really important issue?

Raphael

commit ff52b797 (2016-12-10): Allow configparser to read utf8, fixes #62