ftCommunity/ftcommunity-TXT #63: Playing sounds is really complicate

ski7777, 2016-12-12:
When I will play sound at the moment I have to use the txtsemdcat and mpg123. I have seen that alsa is installed on the TXT but no output device is configured. Could we add the internal speaker to alsa?

Raphael

harbaum, 2016-12-12:
Feel free to write an ALSA-Driver

harbaum, 2016-12-12:
Currently we lack some information (e.g. max spi clock, exact command structure) about the audio interface. But we need to be careful as new information those of us under NDA get must not be published or otherwise used in public unless explicitely stated by the copyright holders. So in order not to taint our setup please make sure you are allowed to use the information in public projects _before_ publishing code.

ski7777, 2016-12-12:
> Feel free to write an ALSA-Driver

Me and writing something so big, no, I to stupid for something like this.

> But we need to be careful as new information those of us under NDA get must not be published or otherwise used in public unless explicitely stated by the copyright holders. So in order not to taint [...]

As long we are only using the data from your MP3-Player I think there is no problem. For all other data I will ask fischertechnik/knobloch before publishing it.

Raphael

ski7777, 2016-12-13:
I just got a relly interesting tip on freenode. We start a background thread, which gets piped via `arecord` to the txt_sebd_cat.If thus is running youncan play audio from everywhere via a simple `aplay` or `mpg123`. Then qou can play multiple sounds simultaneously.
I will try the system in the noon.

Raphael

harbaum, 2016-12-13:
That's true. An external io plugin should be able to do the trick.

http://www.alsa-project.org/alsa-doc/alsa-lib/group___p_c_m___i_o_plug.html

harbaum, 2016-12-15:
BTW: I really like your radio app. Really :-)

Playing around with it got the audio into some "endless repeat loop" sometimes. I am sure that's a problem of my audio client and invoking some program that plays a normal TXT sound through the old firmware can fix this. But that's exaclty of of htese things that need to be debugged before we can use this as a general purpose audio device.