ftCommunity/ftcommunity-TXT #69: Built error

chehr, 2017-01-28:
Due to some built issues a new set-up have been used:

- Oracle VM VirtualBox
- Ubuntu-16.04 LTS.
Unfortunately following error appears during built:
`Download error on https://pypi.python.org/simple/vcversioner/: unknown url type: https -- Some packages may not be found!
Couldn't find index page for 'vcversioner' (maybe misspelled?)
Download error on https://pypi.python.org/simple/: unknown url type: https -- Some packages may not be found!
No local packages or download links found for vcversioner
Traceback (most recent call last):
  File "setup.py", line 44, in <module>
    vcversioner={"version_module_paths" : ["jsonschema/_version.py"]},
  File "/home/christian/ftcommunity-TXT/output/host/usr/lib/python3.5/distutils/core.py", line 108, in setup
    _setup_distribution = dist = klass(attrs)
  File "build/bdist.linux-x86_64/egg/setuptools/dist.py", line 268, in __init__
  File "build/bdist.linux-x86_64/egg/setuptools/dist.py", line 313, in fetch_build_eggs
  File "build/bdist.linux-x86_64/egg/pkg_resources/__init__.py", line 846, in resolve
  File "build/bdist.linux-x86_64/egg/pkg_resources/__init__.py", line 1091, in best_match
  File "build/bdist.linux-x86_64/egg/pkg_resources/__init__.py", line 1103, in obtain
  File "build/bdist.linux-x86_64/egg/setuptools/dist.py", line 380, in fetch_build_egg
  File "build/bdist.linux-x86_64/egg/setuptools/command/easy_install.py", line 633, in easy_install
distutils.errors.DistutilsError: Could not find suitable distribution for Requirement.parse('vcversioner')
`
A manual installation with following commands didn solve the problem:
`sudo apt-get update`
`sudo apt-get install python-vcversioner`
Any idea?

ski7777, 2017-01-28:
https://forum.ftcommunity.de/viewtopic.php?f=33&t=3429&p=27713&hilit=vcversioner#p27734

```
sudo apt-get install python-pip python3-pip
sudo apt-get install python-pip python3-pip
pip install vcversioner
pip3 install vcversioner
```
Then it must work. Other users said, that just executing the python3 commands is enough.

Raphael

chehr, 2017-01-28:
Issue has been solved with the command: `pip3 install vcversioner`
Also final solutions has been discussed on the forum as shown in the link above.
Closed

closed 2017-01-28 by chehr