ftCommunity/ftcommunity-TXT #76: Exploit: bypassing security-check "allow SSH login as ftc" via vnc

olagino, 2017-02-26:
Just determined by laziness of me: The locking-mechanism for SSH-Sessions and root-acces can be bypassed via vnc, so the protection from remote-access is almost useless. Is it a bug, or a feature for lazy programmers? ;)

ski7777, 2017-02-26:
Maybe we cold determine which input device pressed the button and when VNC pressed the button we do just nothing and show a warning and when the Touchscreen presses the OK Button we allow the SSH/Sudo.

Raphael

harbaum, 2017-02-26:
Why not disable VNC completely as long as no password has been set?

ski7777, 2017-02-26:
> Why not disable VNC completely as long as no password has been set?
Maybe. But How to check that?

ski7777, 2017-02-26:
What about this:
`cat /etc/shadow | grep ftc | cut -d: -f2`

rkunze, 2017-02-28:
> Why not disable VNC completely as long as no password has been set?

Actually, I'd prefer to have a GUI to explicitly enable/disable pretty much all of the external connectivity - both basic network services such as WLAN/Bluetooth/USB support as well as higher level services (VNC, HTTP, SSH, ...). And then to default the nonessential services (i.e. at least Bluetooth and VNC) to "disabled".

> Maybe. But How to check that?
> What about this: `cat /etc/shadow | grep ftc | cut -d: -f2`

Works only as root. I'd rather use `/sbin/unix_chkpwd` for this purpose - still a hack (`unix_chkpwd` is internal functionality for PAM and not really meant for general use), but not quite as hacky as grepping /etc/shadow...

harbaum, 2017-03-01:
Since there's no VPN client in the main repository it's definitely a good idea to disable VNC completely. Bluetooth still isn't present at all. And some other services (e.g. OpenVPN) could imho be completely removed.

Having rarely used services eat boot time and memory is indeed a waste.

commit 08df5b41 (2017-05-11): Disable VNC server.

Fixes #76
closed 2017-05-11 by rkunze