ftCommunity/ftcommunity-TXT #78: Add "System Update" app and launcher status bar plugin

rkunze, 2017-03-03:
With the new SD card file system layout introduced in 79e363fc6f8b24b42aa7d2047057c5951620b0fd and the system update script from 8c1f14f4f4ec05cc235e730c54e19afa3eb628b9, we have the necessary functionality for "Over-the-Air" automatic system updates for the community firmware &mdash; we just need a launcher status bar plugin that periodically checks if a new version is available, and a GUI app that displays update metadata (version number etc.) and provides an "Install Update" button.

harbaum, 2017-03-03:
Would it perhaps make sense to integrate the functionality of the about app into this new update app? It's already dealing with system versions and licenses.

ski7777, 2017-03-03:
Some days ago you said it make more sense splitting functionalities is
different apps. So I would prefer to split this into two apps.

Raphael

rkunze, 2017-03-03:
No, I don't think we should integrate "Update" and "About", even if both apps deal with firmware versions and display information &mdash; the functionality is too different: "About" tells the user what is currently installed (i.e. it only displays information), and "Update" installs a new version (i.e it changes the system). I for one would *not* expect the "Update" functionality in something labelled "About"...

Extending the "About" app to indicate if a new version is available (in the "Firmware" line on the "Versions" page) could be a good idea, though.

harbaum, 2017-03-03:
Fine for me ... that was just a suggestion.

harbaum, 2017-03-03:
Such a status bar plugin would import all the qt libraries required to deal with sending http requests. Is the ability to display the update icon really worth the additional load time and memory?

rkunze, 2017-03-03:
Depends - mainly on how much impact the HTTP stuff really has. Do you have any figures for that?

And do we really need to use the Qt HTTP libs? My plan was to use the "native" HTTP support from python (urllib.request).

ski7777, 2017-03-03:
I think urllib.request is enough. We just need one request to the Github
API.

Raphael

ski7777, 2017-07-08:
Can anybody check https://github.com/ftCommunity/ftcommunity-TXT/compare/master...ski7777:update-app ? Just the translation is not working and I don't know why

Raphael

harbaum, 2018-05-15:
Agreed

ski7777, 2018-08-29:
I will update https://github.com/ftCommunity/ftcommunity-TXT/compare/master...ski7777:update-app in the next few days, maybe tonight. Then we have an extended update script that helps to check the update progress. The newly introduced update app can show the progress so the user won't interrupt the update progress because he things it's stuck. In normal mode the app can just update to current stable version. If a snapshot version is installed or the app is started with a special argument you can pick from all versions.

Raphael

oxoovo, 2022-04-08:
[Here](https://github.com/WildCat21/cfw-apps/blob/master/packages/update.zip) is an update app.

ski7777, 2022-12-19:
New discussion here:
Do we need an update app?
Features?
Separate app? Settings app?
...

ski7777, 2022-12-30:
postponing to 1.1

hypnotoad, 2025-01-05:
System updates work reliably via the command line. Maybe it would be time to have a look at this again now.