ftCommunity/ftcommunity-TXT #79: Change documentation to reflect the changes for v0.9.3

rkunze, 2017-03-03:
Version 0.9.3 will have a number of changes from version 0.9.2:
* A new version of ftrobopy that can access the TXT I/O pins without the using the Fischertechnik GUI from original firmware
* Changed look and feel of the launcher
* Changes in some system apps (e.g. WiFi)
* ...

The documentation (especially the wiki) needs to be updated to reflect these changes.

Affected pages (check mark means "updated for 0.9.3"):
* [ ] https://github.com/ftCommunity/ftcommunity-TXT/wiki (image displaying the launcher)
* [ ] https://github.com/ftCommunity/ftcommunity-TXT/wiki/%5BEN%5D-Tutorial%3A-Setting-up-the-ftcommunity-TXT-Firmware
* [ ] https://github.com/ftCommunity/ftcommunity-TXT/wiki/%5BDE%5D-Anleitung%3A-Einrichtung-der-ftcommunity-TXT-Firmware
* [ ] https://github.com/ftCommunity/ftcommunity-TXT/wiki/Tutorial-3%3A-Controlling-a-model (changed init parameters for ftrobopy)
* ...

Note: Documentation/Wiki changes should ideally be made all together and at the same time as the release of v0.9.3 itself in order to not confuse our users. Question: How to best organize this?

ski7777, 2017-03-03:
The wiki is a full git repository. Maybe we could clone this to a full
github repository. After we made all the changes we can merge them in the
wiki repository.

Raphael

rkunze, 2017-03-03:
Yes, we can clone the wiki, make the changes locally (or in a branch) and then merge/push together with the release.

But we need to coordinate this with edits done on the wiki itself, and we have to keep in mind that not everyone who edits the wiki via GitHub is proficient with git, and aware that we are making "shadow" edits to the same pages...

ski7777, 2017-05-30:
I would suggest to close the wiki and move the new documentation into the github page

EstherMi, 2018-03-19:
The documentation has now completely moved to the new structure.
Are there more updates needed for 0.9.3?

rkunze, 2018-03-19:
Mir fallen keine mehr ein - ich mach das Ticket mal zu. Wenn noch irgendwas auftaucht, kann man ja jederzeit noch ein neues Ticket aufmachen.

closed 2018-03-19 by rkunze