ftCommunity/ftcommunity-TXT #80: Sound support in direct mode is unstable

harbaum, 2017-03-03:
The sound support introduced with the direct mode is unstable. After playing a few random  sounds for about half a minute no audio is generated anymore. The txtsndcat based programs then alyo don't work. But starting the original gui via launcher restores audio functionality.

PeterDHabermehl, 2017-03-03:
I think, this is one of the items that must be fixed for a release, thus I added the milestone tag - sorry for not having technical input

rkunze, 2017-03-05:
How about filing an issue against https://github.com/ftrobopy/ftrobopy as well? After all, that's the library that provides the sound support in the first place...

harbaum, 2017-03-05:
Seems to be a local problem for me only. ...

closed 2017-03-05 by harbaum