ftCommunity/ftcommunity-TXT #81: Enable joystick/gamepad kernel modules

PeterDHabermehl, 2017-03-03:
See screenshot - the blue line highlights the missing section.

[image]

rkunze, 2017-03-03:
I'm not sure if these are the right modules &mdash; almost everything under "Joysticks/Gamepads" seems to deal with old hardware connected to a PC gameport or serial port, the sole exception seems to be "X-Box gamepad support".

As far as I can tell from information on the web, support for USB joysticks/gamepads only requires enabling USB HID and joydev ("Joystick Interface" in the screenshot above), and both are already active.

Does anyone know what else is needed for USB joystick/gamepad support?

PeterDHabermehl, 2017-03-03:
Shockingly true, I should have read the <?> help messages...
lsmod on my linux system revealed that on plug-in of the devices the modules hid_dr and ff_memless are loaded... joydev was already loaded before.
Quick research on the net revealed that many of the "cheaper" gamepads require hid_dr, because they're dragonrise clones...

rkunze, 2017-03-03:
OK, that points to the right direction: Device Drivers -> HID support -> HID bus support -> Special HID drivers.

I'll simply enable everything there that looks like a gaming device :smiley:

rkunze, 2017-03-04:
9c963d7bae60caf9021a34cf2c77558f53d09145 adds kernel modules for a number of HID game controllers. 

Anyone with a USB game controller/joystick that did not work out of the box (i.e. with only USB HID and joydev): Please test if it works now, and report here.

ski7777, 2017-03-04:
I can say that my Thrusmaster Flightstick worked without the new modules

PeterDHabermehl, 2017-03-04:
ok-hay...
wireless gamepad now works:

> [   17.823020] usb 1-1: new full-speed USB device number 3 using musb-hdrc
[   17.950493] usb 1-1: New USB device found, idVendor=20bc, idProduct=5500
[   17.950525] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   17.950543] usb 1-1: Product: PC/PS3/Android
[   17.950561] usb 1-1: Manufacturer: ShanWan
[   18.054954] input: ShanWan PC/PS3/Android as /devices/platform/ocp/47400000.usb/47401c00.usb/musb-hdrc.2.auto/usb1/1-1/1-1:1.0/0003:20BC:5500.0002/input/input3
[   18.065181] betop 0003:20BC:5500.0002: input: USB HID v1.10 Gamepad [ShanWan PC/PS3/Android] on usb-musb-hdrc.2.auto-1/input0
[   18.065378] betop 0003:20BC:5500.0002: Force feedback for betop devices by huangbo <huangbobupt@163.com>
[   18.070396] input: ShanWan PC/PS3/Android as /devices/platform/ocp/47400000.usb/47401c00.usb/musb-hdrc.2.auto/usb1/1-1/1-1:1.1/0003:20BC:5500.0003/input/input4
[   18.125029] betop 0003:20BC:5500.0003: input: USB HID v1.01 Device [ShanWan PC/PS3/Android] on usb-musb-hdrc.2.auto-1/input1
[   18.125068] betop 0003:20BC:5500.0003: no output reports found

All three wired gamepads (Logitech + 2x No-Name) DO NOT.
Logitech connects to USB perfectly, but no input device is created. Might be up to an outdated joydev?

>[ 1380.597845] musb-hdrc musb-hdrc.2.auto: VBUS_ERROR in a_wait_bcon (91, <VBusValid), retry #1, port1 00000104
[ 1381.192523] usb 1-1: new full-speed USB device number 5 using musb-hdrc
[ 1381.321468] usb 1-1: New USB device found, idVendor=046d, idProduct=c21d
[ 1381.321503] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1381.321521] usb 1-1: Product: Gamepad F310
[ 1381.321538] usb 1-1: Manufacturer: Logitech
[ 1381.321554] usb 1-1: SerialNumber: 9B33CAEF
[ 1413.651355] usb 1-1: USB disconnect, device number 5

Very cheap gamepad does not connect to the USB stack properly, poor implementation on the gamepad, I guess... Basically same happens with the "better" no-name-gamepad.

>[ 1424.103640] usb 1-1: new low-speed USB device number 6 using musb-hdrc
[ 1429.230206] usb 1-1: device descriptor read/all, error -71
[ 1429.343784] usb 1-1: new low-speed USB device number 7 using musb-hdrc
[ 1434.470295] usb 1-1: device descriptor read/all, error -71
[ 1434.583913] usb 1-1: new low-speed USB device number 8 using musb-hdrc
[ 1434.600359] usb 1-1: device descriptor read/8, error -71
[ 1434.725257] usb 1-1: device descriptor read/8, error -71
[ 1434.943911] usb 1-1: new low-speed USB device number 9 using musb-hdrc
[ 1434.960352] usb 1-1: device descriptor read/8, error -71
[ 1435.085358] usb 1-1: device descriptor read/8, error -71
[ 1435.194084] usb usb1-port1: unable to enumerate USB device

on the Linux-PC, dmesg reports:
>[20728.525145] usb 8-1.3: new low-speed USB device number 8 using ohci-pci
[20728.643441] usb 8-1.3: New USB device found, idVendor=0079, idProduct=0011
[20728.643455] usb 8-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[20728.643457] usb 8-1.3: Product: USB Gamepad 
[20729.065389] input: USB Gamepad  as /devices/pci0000:00/0000:00:13.0/usb8/8-1/8-1.3/8-1.3:1.0/0003:0079:0011.0007/input/input23
[20729.065573] dragonrise 0003:0079:0011.0007: input,hidraw5: USB HID v1.10 Joystick [USB Gamepad ] on usb-0000:00:13.0-1.3/input0

so I see ohci-pci being used instead of musb-hdrc. Anyone to draw conclusions?

Thanks
Peter

rkunze, 2017-03-04:
A bit of googling tells me that the Logitech Gamepad F310 supports two modes: An xbox compatible xinput" mode and a "direct" mode. There should be a little switch at the rear to change the mode. Can you try again in "direct" mode? Or try again with 1a84cfcd39b9c1c4f0fa977aef4561ae2b70bd0c (this adds the driver for xbox compatible gamepads, which should work with the Logitect F310 in xinput mode).

Regarding the other two controllers: The error messages indicate a problem with initialising the device - my guess would be a real hardware incompatibility here. Apart from  that, the controller with usb id 0079:0011 should in theory be supported by the hid_dr controller. And the ohci-pci/musb-hdrc difference is just the drivers for different host controllers on the PC and the TXT.

harbaum, 2017-03-04:
The f310 is nasty. On my desktop pc running linux it only works with the xbox driver and in xbox mode. On my mist board it works only in regular mode. Never tried on the txt.

harbaum, 2017-03-04:
Tested and it works in neither mode. Its mode LED doesn't even work and that's something the controller does all by itself. 

Actually many of my joysticks don't work. Some work when connected via a self powered usb hub.

PeterDHabermehl, 2017-03-04:
:-(

Thought, the Logitech is a a quality brand product...
I don't even have a powered usb hub (only built-in on my display, which is acually not very handy at the moment)...

rkunze, 2017-03-05:
> Actually many of my joysticks don't work. Some work when connected via a self powered usb hub.

This could be the related to the problem we had with the ROBO LT Controller &ndash; apparently, the TXT USB controller is a bit finicky regarding timings and power consumption by USB clients.

rkunze, 2017-03-05:
I'm closing this issue now as generic joystick/gamepad support is enabled now and I think that support/debugging of individual joystick/gamepad models is better handled in separate issues (one per model/chip/whatever).

closed 2017-03-05 by rkunze

harbaum, 2017-03-05:
That may definitely be related. But it's usually the devices that expect a usb power source to tollerate if they exceed the spec.