ftCommunity/ftcommunity-TXT #82: Remove unused 'applog' feature from launcher

harbaum, 2017-03-04:
The launcher can redirect an apps stdout and stderr output to a web page. This feature doesn't seem to be used. I'd like to remove it.

PeterDHabermehl, 2017-03-04:
I didn't know about the feature until now. How is it used? Can it be used for webinterface script output, too?
Then I would definitely like to keep the feature...

ski7777, 2017-03-04:
When a feature is not in a official release, nobody will use it. The developers usually don´t use these kinds of features because they know how to use the shell but normal users will need to use it to generate error reports.

Rapahel

harbaum, 2017-03-04:
It's been in the official release since quite some time now after you asked for it if i remember correctly. Still noone has used it.

harbaum, 2017-03-04:
No,  this feature only works for apps and is built into the launcher. It'll also not work with the new light apps.

harbaum, 2017-03-09:
It seems some people are still using it or at least would like to use it. So we just keep it ....

closed 2017-03-09 by harbaum