ftCommunity/ftcommunity-TXT #86: Network error

chehr, 2017-04-14:
The error message " Netzwerkfehler: Verbindung verweigert" can not be left by pressing "X" or via Network Browser. It seem that the only possibility is to reboot the TXT.

Note: The error message itself is based on a filter which has been set manually by the user with the Fritzbox in order that the TXT is not communicating with the Internet. So, the error message is correct.

ski7777, 2017-04-15:
Same when TXT has no WLAN.

ski7777, 2017-04-15:
Here is the Traceback:
```
Traceback (most recent call last):
  File "/opt/ftc/TouchStyle.py", line 291, in close
    self.parent.unregister(self)
AttributeError: 'QWidget' object has no attribute 'unregister'
```

harbaum, 2017-04-15:
You didn't even mention which application is having this problem. The store?

ski7777, 2017-04-15:
Store

harbaum, 2017-04-15:
Argh ... closed by accident ...

Anyway,  did this happen with the old non-light version as well?

PeterDHabermehl, 2017-04-26:
Can confirm the issue on tx-pi.

harbaum, 2017-05-17:
The bug is in TouchMessageBox ....

closed 2017-05-17 by harbaum

PeterDHabermehl, 2017-05-17:
Did I induce the problem when "updating" the message box? hope not... if so, sorry...

commit d0fa85f0 (2017-05-17): Don't set self.parent in TouchMessageBox. Fixes #86