ftCommunity/ftcommunity-TXT #95: Update python tutorial app

rkunze, 2017-05-28:
Currently, the links to the python tutorial app on http://cfw.ftcommunity.de/ftcommunity-TXT/en/programming/python/tutorial-1.html and http://cfw.ftcommunity.de/ftcommunity-TXT/de/programming/python/tutorial-1.html are broken because the old tutorial app is no longer in the main app repository at https://github.com/ftCommunity/ftcommunity-apps, and the old tutorial app at https://github.com/harbaum/cfw-apps/tree/master/packages/app_tutorial_1 does not match the tutorial any more.

Updating the app to match the current tutorial and re-adding it to https://github.com/ftCommunity/ftcommunity-apps is an easy task for a novice contributor.

harbaum, 2017-05-29:
I'd suggest setting up a repository especially for those tutorials and similar examples. Placing them in the main app store imho doesn't make too much sense since the majority of users has no use for them.

rkunze, 2017-05-29:
Good point. And on reflection, installing the tutorial apps via the app store is pretty pointless anyway - after all, the tutorials are about writing, packaging and uploading an app _without_ using the app store.

In this case, it's probably better to simply add the sources (test.py, icon and the manifest) and maybe the zip file directly to cfw.ftcommunity.de (i.e. place them somewhere below `docs` in the https://github.com/ftCommunity/ftcommunity-TXT repository), or do away with the download links for the tutorial sources altogether (the whole whopping 17 lines of source code are shown on the tutorial site anyway, ready for copy-and-paste).

ski7777, 2017-05-30:
I would suggest to enable an extra github page for the apps repository.
Later we should open a new repository for the demo and test apps. There I would suggest to open also a github page.

LarsK1, 2017-05-30:
I would suggest we create a File Tree like that:
docs/resources (for everything in neutral language)
docs/resources/en(for everything special in english)
docs/resources/...

rkunze, 2017-05-30:
We already have `docs/media/` for content that does not depend on the language.

I have an idea how to handle the tutorial apps so that we have both download links and up-to-date content in the pages (via includes) - I'll try it this evening or tomorrow.

LarsK1, 2017-06-21:
Do you merged already?

rkunze, 2017-06-24:
Finally got around to do it - see commit 69bbaaaceb32d6c5de1663b192dcaacf92a7baf4.

Example files can now be added below the `docs/_includes/examples/` directory (or a subdirectory), and included in a markdown page with `{% include examples/whatever % }`. In addition, all files and directories below `docs/_includes/examples/` will be copied to `media/examples/...` when the site is deployed so the can be linked to as well.

See also the python tutorial-1.md files for an example - the python source code and manifest in there is both included in the page and linked for download.

closed 2017-06-24 by rkunze