ftCommunity/ftcommunity-TXT #119: Error while booting on advanced layout

OSevangelist, 2018-05-12:
Since a couple of weeks (in more or less a few iterations) i tried to make the Community firmware run on the advanced layout. However, after a fresh build of the most recent RC tag (but also all prior versions i tried) and after following the exact approach to create the device layout as

```
parted /dev/mmcblk0 mklabel msdos
parted -a optimal /dev/mmcblk0 -- mkpart primary fat32 1MB -300MB
parted -a optimal /dev/mmcblk0 -- mkpart primary ext2 -300MB 100%
parted /dev/mmcblk0 set 1 boot on
mkfs.vfat -n BOOT /dev/mmcblk0p1
mkfs.ext4 -L ROOT /dev/mmcblk0p2
```
and copy the files as files `output/images/uImage` and `output/images/am335x-kno_txt.dtb` to the first partition of the SD card:

```
mount /dev/mmcblk0p1 /mnt
cp output/images/uImage output/images/am335x-kno_txt.dtb /mnt/
umount /mnt
```

Finally, unpack `output/images/rootfs.tar` to the second partition on the SD card:

```
mount /dev/mmcblk0p2 /mnt
tar xvf output/images/rootfs.tar -C /mnt/
umount /mnt
```
Booting does not complete. The progress bar just remains at the beginning with the blue led continously lighting. I have read through many links / tipps and tricks and also came across this

https://forum.ftcommunity.de/viewtopic.php?f=33&t=3903

that effectively suggest to have a rootfs.conf file in the FAT partition with content rw but also this didn't change anything. Does anyone have an idea what may went wrong at my side ? Is it right that i have only 

am335x-kno_txt.dtb
uImage
rootfs.conf

on the BOOT partition ?

rkunze, 2018-05-12:
Since release 0.9.3, we don't have the distinction between simple and advanced layout anymore. Instead, we're generally using a partition scheme similar to the advanced layout (i.e. one FAT partition used for booting, and an ext4 partition for user data).

The difference to the old advanced layout is that we're now always using the rootfs.img as the main, read-only system image, and mount specific directories on the ext4 partition to those parts of the file system tree that should be writeable (mainly /etc and /home).

The partitioning is done automatically on first boot, and that's what causes the boot to fail in your case: You already have a second ext4 partition (so the first boot setup code won't run), but you don't have the necessary directory structure on the ext4 partitions which the rest of the setup assumes to be there.

Easiest solution: Repartition the SD card to a single, large FAT partition, dump am353x-kno_txt.dtb. uImage and rootfs.img on it, and reboot. After the first boot, you'll have a FAT partition of around 500MB, and an ext4 partition using the remaining space of the SD card.

>  I have read through many links / tipps and tricks and also came across this
>
> https://forum.ftcommunity.de/viewtopic.php?f=33&t=3903
>
> that effectively suggest to have a rootfs.conf file in the FAT partition with content rw but also this didn't > change anything.

rootfs.conf with content "rw" just gives you a writable root (as opposed to the usual readonly root). Technically, this is done by mounting another part of the ext4 filesystem as a writeable overlay over the read-only image file.

OSevangelist, 2018-05-13:
Hi @rkunze thanks for that quick response. Your suggestion worked out well at my side. I don't know what exactly the verified "tag" besides the build means, but i can confirm release
https://github.com/ftCommunity/ftcommunity-TXT/tree/v0.9.4-dev.snapshot-20180309
worked for me and i would hence "verify" that this is running on my RoboTXT [1].
 
WLAN configuration however, by just dropping the wpa_supplicant.conf file on the created overlay filesystem in the second partition under /overlay/etc/wpa_supplicant.conf didn't work out for me so i had to configure WLAN manually. 

Furthermore, i could be mislead but wasn't there a package manager available in earlier versions of the firmware and wouldn't it be helpful to have such a thing onboard to easyinstall further packages on the RobotTXT

Finally may i suggest to slightly change the README.md in order to reflect the design change of creating the advanced layout "on the fly"

Thanks a lot and after all great effort and great work to bring real Open Source to this piece of hardware

[1] The latest release after the actual built complaints about a mismatch while doing "git describe".

OSevangelist, 2018-05-13:
@rkunze ok obviously i accidentally thought that a package manager was available when i tried the community firmware a couple of weeks ago but it seems that you've taken the design decision to use buildroot as a solid foundation for the work. Henceforth i found a reasonable answer on my request here http://buildroot.uclibc.org/downloads/manual/manual.html#faq-no-binary-packages
that effectively says "buildroot" in a attempt to focus on the most necessary parts of an embedded Linux system doesn't provide a binary package manager. However i was asking myself whether python based development would'nt be much easier and more convenient if at least PIP and VIRTUALENV where available of the resulting image? As far as i understand it should be fairly easy to integrate these according to section 

17.8. Infrastructure for Python packages ff.

in https://buildroot.org/downloads/manual/manual.html#_infrastructure_for_python_packages 
But i was wondering what you guys think of it or whether there is good reason to not integrate it on the device?

rkunze, 2018-05-16:
We've decided against a packaging system for two reasons:

* Limited resources on the target as mentioned by @PeterDHabermehl - mainly CPU and RAM, and especially the CPU and RAM needed to run a C Compiler. That may be possible, but - apart from trivial programs - probably dead slow and prone to running out of RAM. Unfortunately, for full PIP support on the TXT, a C compiler is needed. Otherwise, you can only install pure python packages (without any native code), which makes PIP pretty much useless.
* A smaller environment to support. As its stands, we have _one_ current version of the CFW, including everything from kernel to the Python environment, and do not need to worry about supporting a zillion different environments caused by a zillion combinations of packages.

OSevangelist, 2018-05-17:
Hi @PeterDHabermehl and @rkunze thanks for analyzing that matter in depth. Honestly i wasn't aware of the dependency to C and a C compiler to use PIP, but owing that fact and the limited resources of the TXT that really doesn't make too much sense then. Thanks for clarifying this