ftCommunity/ftcommunity-TXT #121: Remove wiki and update Readme

EstherMi, 2018-06-17:
All relevant content of the wiki has been copied to the documentation. The wiki is not updated any more (would be double work) and is now obsolete.
The readme also needs some update to keep track with the documentation.

rkunze, 2018-06-17:
Cleaned up README.md in ed557ec623143e37a1a11b6b3a0845149968630e and 138139f32e89320922359801f421ccee67895120, moved the serial console access page to the web site, and hacked up a german translation.

What's still left are the images in https://github.com/ftCommunity/ftcommunity-TXT/wiki/%5BEN%5D-Images - do we still need those? If not, I'll turn off the wiki in the project settings.

harbaum, 2018-06-18:
Most of the images are from me. I don't need then anymore.

rkunze, 2018-06-21:
I have now switched off the wiki in the project settings.

Contents are still available by cloning https://github.com/ftCommunity/ftcommunity-TXT.wiki.git

closed 2018-06-21 by rkunze