ftCommunity/ftcommunity-TXT #122: clock set to 2000-01-01 upon boot

ftrobopy, 2018-07-07:
The hardware clock of the TXT is always initialised to 0 ( = Jan 1st 2000 00:00) upon booting cfw 0.9.4 resulting in a system time of Jan 1st 1970 01:00 am after each boot.
This behaviour implies an empty battery. However the battery is not the problem.
Using the same TXT with the original firmware (or older cfw-version 0.9.3) does not show this problem.

harbaum, 2018-07-07:
What about this one? https://github.com/ftCommunity/ftcommunity-TXT/pull/104

ftrobopy, 2018-07-07:
it seems that #104 is not working in 0.9.4

ftrobopy, 2018-07-07:
I used 0.9.4-dev+snapshot-20171215 before ... the clock was working correctly in that version.

ftrobopy, 2018-07-09:
I rechecked version 0.9.4-dev+snapshot-20171215 ... correction: the clock does _not_ work in this version ... I only thought it was working but upon closer looking I noticed that the clock was set from the network by ntpd  (not by reading the hardware clock of the TXT).
Putting back the line (see also #104)
CONFIG_RTC_DRV_OMAP=y
into the file "board/fischertechnik/TXT/tisdk_am335x-fischertechnik_txt_defconfig" did help.

The system time is now correctly initialised from the hardware clock upon boot. However the hardware clock is not synchronised to the system time when shutting down.
Adding the line
CONFIG_RTC_SYSTOHC=y
would help keeping system time and hardware clock synchronised: (from the Linux Kernel Driver Database) "[...] the system time (wall clock) will be stored in the RTC specified by RTC_HCTOSYS_DEVICE approximately every 11 minutes if userspace reports synchronized NTP status [...]"

ftrobopy, 2018-07-09:
starting the ft-GUI app and setting the time within the app does synchronize system time and hardware clock correctly (only if CONFIG_RTC_DRV_OMAP=y).

ski7777, 2018-07-16:
This is what I found out in the schematics: (According to my NDA I'm allowed to publish information - not the full schematics - for open source software development)
- There is a CPU and a power control chip (TPS)
- There is obviously a backup battery
- Both CPU and TPS are connected to the backup battery
- The wake-up output of the TPS is connected to the power-on circuit to allow wake on RTC

The RTC in the CPU seems to be resetted after every power down although it's connected to battery power, too. #104 was an attempt to disable this buggy CPU RTC.

After my holiday I will try to investigate more. I think it shouldn't be that complicated.

Raphael

rkunze, 2018-09-15:
Fixed in c5bdb958918d8f8167b22004a49ba86aa990f4d3.
544efba40ae88965df1f84d969579592e4643797 sets CONFIG_RTC_SYSTOHC=y as suggested to automatically sync NTP time back to rtc1.

Side note on RTC behaviour: It looks like rtc0 is stopped when the TXT is powered off, but keeps the time of the last shutdown and starts again from this time at boot. So you can tell how long the TXT has been powered off by comparing rtc0 and rtc1 (which keeps running when powered off) :smile:

closed 2018-09-15 by rkunze