ftCommunity/ftcommunity-TXT #143: Add a shutdown popup

ski7777, 2018-12-09:
- [ ] Add a popup window
    - [ ] Add shutdown
    - [ ] Add reboot
    - [ ] Add screenshot
    - [ ] Add debug log generation
- [ ] TouchStyle: Don't close the app on long button press
- [ ] Triggerhappy: Distinguish between short, middle and long press
- [ ] Multitasking: Add app switcher

Button press lengths:
- <2s: close app
- 2s-5s: open popup
- \>5s: force shutdown

ski7777, 2023-06-29:
@hypnotoad What do you think about the idea of a popup?

hypnotoad, 2025-01-05:
I am closing this issue as there is no progress and my focus is on maintainance currently (https://github.com/ftCommunity/ftcommunity-TXT/discussions/297).

This does not mean that the ticket should not be done, but it is rather to declutter the issue list.

closed 2025-01-05 by hypnotoad