ftCommunity/ftcommunity-TXT #144: Add multitasking support

ski7777, 2018-12-09:
This is just a proposal. I know that the TXT's resources are very limited. So we should try to make this as lightweight as possible. Maybe we will try this and decide that this should not be enabled by default.

- [ ] Allow running of multiple apps and introducing a mechanism to switch between the apps
- [ ] Integrate this in the power popup #143

WildCat21, 2021-06-21:
Is this possible?

ski7777, 2022-12-19:
@hypnotoad Probably you are most familiar with the new Qt5/X11 solution. Is there a chance of implementing this?

hypnotoad, 2022-12-19:
I am in turn not familiar with touch usability, but if someone has a concept or some concrete idea, I think we should discuss it.

For intance, we could extend the top bar with a "minimize" icon and re-open an app when the launcher launches it again.

ski7777, 2022-12-20:
I will put that on hold at least until v1.0.0 is out

ski7777, 2023-06-29:
@hypnotoad You implemented these X11 features. Do you think it is possible to hid/show apps?

hypnotoad, 2024-10-04:
I am closing this suggestion as I don't think it is realistic (hardware and software-wise). If someone would like to help, there are other topics helping more.