ftCommunity/ftcommunity-TXT #146: Enhance security

ski7777, 2018-12-09:
Currently all apps and the launcher run with the same rights. This can cause devastating security issues.
Proposal:
Create a new user account/group per app
Save the app files with ftc/ftc permissions and create app data folder with the app permissions
Run app with this permissions

If an app needs to access data of another app, either the developer of the accessed app has to allow this in the manifest (the user will be asked during installation) or the app can request it during run (here we should introduce a system where the user can revoke these permissions)

ski7777, 2022-12-30:
postponing to 1.1

ski7777, 2023-01-16:
Just a weird proposal: Can we run apps in containers?

hypnotoad, 2025-01-05:
I am closing this issue as there is no progress and my focus is on maintainance currently (https://github.com/ftCommunity/ftcommunity-TXT/discussions/297).

This does not mean that the ticket should not be done, but it is rather to declutter the issue list.

closed 2025-01-05 by hypnotoad