ftCommunity/ftcommunity-TXT #152: USB broken

ski7777, 2018-12-13:
On https://github.com/ftCommunity/ftcommunity-TXT/tree/v0.9.5-rc%2Bsnapshot-20180916 it wokrs. USB is an issue on master only.

Both USB ports are not working. When connecting the TXT to a PC it won´t react. Connecting USB devices to the TXT does not work, too.

I tried a `git bisect` but I couldn´t find the bad commit because the were too many states which were not bootable.
I´m quite sure the problem is somewhere in the DTS or the kernel config. @rkunze Do you know how to fix that?

`lsusb` prints out nothing and return code is 1

Boot log:
```
[    0.000933] timer_probe: no matching timers found
[    1.419491] mtdoops: mtd device (mtddev=name/number) must be supplied
[    1.440988] spidev spi1.0: buggy DT: spidev listed directly in DT
[    1.747455] omap_voltage_late_init: Voltage driver support not added
mv: can't rename '/media/sdcard/data/overlay/rootfs': Directory not empty
UBI device number 0, total 964 LEBs (124379136 bytes, 118.6 MiB), available 0 LEBs (0 bytes), LEB size 129024 bytes (126.0 KiB)
Starting logging: OK
Starting thd: ␛[?25l␛[H␛[Jfbv - The Framebuffer Viewer
/etc/ftc-logo.png
240 x 320
OK
␛[H␛[J␛[?25hPopulating /dev using udev: [   12.195002] wlcore: ERROR could not get configuration binary ti-connectivity/wl18xx-conf.bin: -2
done
Initializing random number generator... done.
Starting rngd: OK
Starting system message bus: done
Starting network: ip: can't find device 'usb0'
ip: SIOCGIFFLAGS: No such device
Successfully initialized wpa_supplicant
udhcpc: started, v1.29.3
udhcpc: sending discover
ip: SIOCGIFFLAGS: No such device
ip: can't find device 'bnep0'
ip: SIOCGIFFLAGS: No such device
FAIL
Starting ntpd: OK
Starting lighttpd: OK
Starting sshd: /etc/ssh/sshd_config line 109: Deprecated option UsePrivilegeSeparation
OK
Starting bluetooth: OK
Starting DHCP server: FAIL
Starting launcher: OK

Welcome to FT-txt
FT-txt login: udhcpc: sending discover
```

`dmesg`:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.87 (raphael@raphael-laptop) (gcc version 8.2.1 20180802 (GNU Toolchain for the A-profile Architecture 8.2-2018-08 (arm-rel-8.23))) #24 PREEMPT Thu Dec 13 12:21:43 CET 2018
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Knobloch TXT AM3352
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 16 MiB at 0x8e000000
[    0.000000] On node 0 totalpages: 65024
[    0.000000] free_area_init_node: node 0, pgdat c0ede3a4, node_mem_map cfbf1000
[    0.000000]   Normal zone: 508 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65024 pages, LIFO batch:15
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (neon)
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64516
[    0.000000] Kernel command line: fbtft_device.name=txt_ili9341 fbtft_device.fps=10 console=ttyO0,115200 root=/dev/mmcblk0p2 rw rootwait quiet
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[... 342 lines cut ...]
[   12.611806] wlcore: wl18xx HW: 183x or 180x, PG 2.2 (ROM 0x11)
[   12.766635] wlcore: loaded
[   13.357623] udc-core: couldn't find an available UDC - added [g_ether] to list of pending drivers
[   14.642252] wlcore: PHY firmware version: Rev 8.2.0.0.242
[   14.750068] wlcore: firmware booted (Rev 8.9.0.0.79)
[   14.789600] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   17.071288] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.071374] Bluetooth: BNEP socket layer initialized
[   17.185607] Bluetooth: RFCOMM TTY layer initialized
[   17.185694] Bluetooth: RFCOMM socket layer initialized
[   17.185796] Bluetooth: RFCOMM ver 1.11
```

`dmesg | grep usb`:
```
[    0.706295] usbcore: registered new interface driver usbfs
[    0.707239] usbcore: registered new interface driver hub
[    0.710945] usbcore: registered new device driver usb
[    1.463108] usbcore: registered new interface driver cdc_acm
[    1.463818] usbcore: registered new interface driver cdc_wdm
[    1.464577] usbcore: registered new interface driver usb-storage
[    1.465023] usbcore: registered new interface driver usbtest
[    1.567903] usbcore: registered new interface driver usbhid
[    1.567923] usbhid: USB HID core driver
```

rkunze, 2018-12-13:
> I'm quite sure the problem is somewhere in the DTS or the kernel config. 

Yes, probably. And since the boot log does not show anything obvious regarding USB, my first guess is the DTS.

> @rkunze Do you know how to fix that?

Not out of hand. I'll try to find it this evening.

If you want to give it a try yourself: Look through anything containing "usb" in the DTS, compare with the corresponding sections from similar devices (beaglebone or the TI evaluation boards looks like the closest "cousins" to me), and see if anything catches your eye...

And yes, we should have tested it.

rkunze, 2018-12-13:
Turns out it was the kernel config after all.

commit c9a13109 (2018-12-13): Fix USB config for kernel 4.14

Closes #152
closed 2018-12-13 by rkunze