ftCommunity/ftcommunity-TXT #162: Keep auto-categorization of apps even if an app was moved

heuer, 2019-03-24:
Once an app is moved, all later installations ignore the "category" in the manifest and the apps are placed on the homescreen.

Would be nice if the auto-categorization can be kept.

See <https://github.com/ftCommunity/tx-pi/issues/27>

> [...] There's one minor disadvantage: Once apps are moved the config file storing app location info is created and further apps installed later are not automatically store in their default categories anymore.
>
> Bascially once you start moving stuff around any auto-categorization is disabled.

ski7777, 2023-06-29:
I guess fixing this should not be too complicated but in my opinion we could remove auto-categorization completely

ski7777, 2023-06-29:
@hypnotoad Any coments=´?

hypnotoad, 2025-01-05:
I am closing this issue as there is no progress and my focus is on maintainance currently (https://github.com/ftCommunity/ftcommunity-TXT/discussions/297).

This does not mean that the ticket should not be done, but it is rather to declutter the issue list.

closed 2025-01-05 by hypnotoad