ftCommunity/ftcommunity-TXT #166: State of harbaum/TouchUI vs TXT/TouchUI?

heuer, 2019-04-07:
For a newbie like me it's unclear which repository is mandatory.

The file ``TouchStyle.py`` is even with <https://github.com/harbaum/TouchUI/blob/master/touchui/TouchStyle.py> but ``launcher.py`` contains a lot of changes which have not been migrated to <https://github.com/harbaum/TouchUI/blob/master/touchui/launcher.py>.
Maybe by intention, but this wasn't clear, at least for me.

I modified <https://github.com/harbaum/TouchUI/blob/master/touchui/launcher.py> since I thought the changes can be easily applied to the ``ft-TXT`` revision but I was mistaken.

I try to incoperate my changes to  <https://github.com/harbaum/TouchUI/blob/master/touchui/launcher.py> to the ``ft-TXT`` version but it would have been helpful to know that TouchUI is not ``ft-TXT`` TouchUI in advance.

Maybe it's my fault, but I couldn't find any document which would have told me the differences. In the future, any modification to ``TouchStyle.py`` should go to ``harbaum/TouchUI`` and modifications to ``launcher.py`` should be applied to this repository, I guess?

hypnotoad, 2022-06-12:
@harbaum : What is your opinion here? I have done lots of changes during the Qt5 migration in the ftcommunity-TXT branch. How active do you want to be in future development of launcher or touchstyle?

harbaum, 2022-06-12:
TouchUI started as a standalone project. Development then continued in the txt firmware and became somewhat txt specific. I don't see a reason to backport it. Just ignore the separate repo.

ski7777, 2022-12-19:
I think we can close this here

closed 2022-12-19 by ski7777