ftCommunity/ftcommunity-TXT #168: TouchUI / Launcher: Move BusyAnimation to TouchStyle?

heuer, 2019-04-09:
The BusyAnimation is used by a few apps. But they don't use the BusyAnimation class from the launcher module but use copy of it. 

IMO it would be nice if the BusyAnimation class would be part of TouchStyle and other apps could simply import that class without copy and paste.

Thoughts?

heuer, 2019-04-11:
These (system) apps use (a copy of) BusyAnimation:

* bluetooth
* store
* TX-Pi config 

I thought there were more. But I am still in favor of moving the BusyAnimation to TouchStyle

hypnotoad, 2022-06-05:
I think it would be a good idea

ski7777, 2022-12-19:
bump
@hypnotoad Can you do this? Would be really nice :)

ski7777, 2022-12-30:
Any updates here @hypnotoad

hypnotoad, 2022-12-31:
The code of the launcher and the other modules is different. I will shift this to 1.1.

ski7777, 2022-12-31:
ok

closed 2023-05-23 by ski7777