ftCommunity/ftcommunity-TXT #170: TXT Firmware: git clone ... + make FAILED ?

bo911, 2019-05-06:
Try to build TXT firmware ...
devel@ub64:/opt/FTC$ git clone https://github.com/ftCommunity/ftcommunity-TXT.git
Klone nach 'ftcommunity-TXT' ...
remote: Enumerating objects: 108, done.
remote: Counting objects: 100% (108/108), done.
remote: Compressing objects: 100% (70/70), done.
remote: Total 282605 (delta 40), reused 58 (delta 17), pack-reused 282497
Empfange Objekte: 100% (282605/282605), 64.76 MiB | 24.49 MiB/s, Fertig.
Löse Unterschiede auf: 100% (193956/193956), Fertig.
Prüfe Konnektivität ... Fertig.
devel@ub64:/opt/FTC$ cd ftcommunity-TXT
devel@ub64:/opt/FTC/ftcommunity-TXT$ make
git submodule update --init buildroot
Submodul 'buildroot' (https://github.com/ftCommunity/buildroot) für Pfad 'buildroot' in die Konfiguration eingetragen.
Klone nach 'buildroot' ...
remote: Enumerating objects: 312641, done.
remote: Total 312641 (delta 0), reused 0 (delta 0), pack-reused 312641
Empfange Objekte: 100% (312641/312641), 85.66 MiB | 25.41 MiB/s, Fertig.
Löse Unterschiede auf: 100% (208213/208213), Fertig.
Prüfe Konnektivität ... Fertig.
fatal: Referenz ist kein "Tree"-Objekt: 2083e090a70da9fa50dfcc0179a9a53e3e10a355
Konnte '2083e090a70da9fa50dfcc0179a9a53e3e10a355' in Submodul-Pfad 'buildroot' nicht auschecken.
Makefile:9: die Regel für Ziel „buildroot/Makefile“ scheiterte
make: *** [buildroot/Makefile] Fehler 1

heuer, 2020-05-07:
I get another error. Cloning the submodule buildroot works, but...
```
make
git submodule update --init buildroot
Submodul 'buildroot' (https://github.com/ftCommunity/buildroot) für Pfad 'buildroot' in die Konfiguration eingetragen.
Klone nach '/home/tacc/ftcommunity-TXT/buildroot' ...
Submodul-Pfad: 'buildroot': '2083e090a70da9fa50dfcc0179a9a53e3e10a355' ausgecheckt
BR2_EXTERNAL=.. make -C buildroot fischertechnik_TXT_defconfig
make[1]: Verzeichnis „/home/tacc/ftcommunity-TXT/buildroot“ wird betreten
mkdir -p /home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config/lxdialog
PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \
    obj=/home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config -C support/kconfig -f Makefile.br conf
make[2]: Verzeichnis „/home/tacc/ftcommunity-TXT/buildroot/support/kconfig“ wird betreten
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config -DCONFIG_=\"\"  -MM *.c > /home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config/.depend 2>/dev/null || :
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config -DCONFIG_=\"\"   -c conf.c -o /home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config/conf.o
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config -DCONFIG_=\"\"  -I. -c /home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config/zconf.tab.c -o /home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config/zconf.tab.o
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config -DCONFIG_=\"\"   /home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config/conf.o /home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config/zconf.tab.o  -o /home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config/conf
rm /home/tacc/ftcommunity-TXT/buildroot/output/build/buildroot-config/zconf.tab.c
make[2]: Verzeichnis „/home/tacc/ftcommunity-TXT/buildroot/support/kconfig“ wird verlassen

WARNING: unmet direct dependencies detected for BR2_PACKAGE_BLUEZ_UTILS
  Depends on [n]: !BR2_STATIC_LIBS [=n] && BR2_USE_WCHAR [=n] && BR2_TOOLCHAIN_HAS_THREADS [=y] && BR2_USE_MMU [=y]
  Selected by [y]:
  - BR2_PACKAGE_FT_BT_REMOTE_SERVER [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PYUDEV
  Depends on [n]: (BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=n]) && BR2_PACKAGE_HAS_UDEV [=n]
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_FTTXPY [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_SERIAL
  Depends on [n]: BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=n]
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_FTROBOPY [=y]
  - BR2_PACKAGE_PYTHON_FTTXPY [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PILLOW
  Depends on [n]: BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=n]
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_LUMACORE [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PYUSB
  Depends on [n]: (BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=n]) && BR2_TOOLCHAIN_HAS_THREADS [=y]
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_ROBOLT [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_BINUTILS
  Depends on [n]: !BR2_nios2 [=n] && BR2_USE_WCHAR [=n]
  Selected by [y]:
  - BR2_PACKAGE_LIBROBOINT [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_BLUEZ_UTILS
  Depends on [n]: !BR2_STATIC_LIBS [=n] && BR2_USE_WCHAR [=n] && BR2_TOOLCHAIN_HAS_THREADS [=y] && BR2_USE_MMU [=y]
  Selected by [y]:
  - BR2_PACKAGE_FT_BT_REMOTE_SERVER [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PYUDEV
  Depends on [n]: (BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=n]) && BR2_PACKAGE_HAS_UDEV [=n]
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_FTTXPY [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_BINUTILS
  Depends on [n]: !BR2_nios2 [=n] && BR2_USE_WCHAR [=n]
  Selected by [y]:
  - BR2_PACKAGE_LIBROBOINT [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PILLOW
  Depends on [n]: BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=n]
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_LUMACORE [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PYUSB
  Depends on [n]: (BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=n]) && BR2_TOOLCHAIN_HAS_THREADS [=y]
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_ROBOLT [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_SERIAL
  Depends on [n]: BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=n]
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_FTROBOPY [=y]
  - BR2_PACKAGE_PYTHON_FTTXPY [=y]
#
# configuration written to /home/tacc/ftcommunity-TXT/buildroot/.config
#
make[1]: Verzeichnis „/home/tacc/ftcommunity-TXT/buildroot“ wird verlassen
make -C buildroot
make[1]: Verzeichnis „/home/tacc/ftcommunity-TXT/buildroot“ wird betreten
Makefile:569: *** check is in the dependency chain of bluez_utils that has added it to its _DEPENDENCIES variable without selecting it or depending on it from Config.in.  Schluss.
make[1]: Verzeichnis „/home/tacc/ftcommunity-TXT/buildroot“ wird verlassen
make: *** [Makefile:3: all] Fehler 2
```

heuer, 2020-05-11:
Any idea what's going wrong or how to fix it?
It's a bit difficult to participate when even the basic things don't work.

Any hints are appreciated.

heuer, 2020-05-12:
No worries; works now.

heuer, 2020-07-02:
Since there was no feedback from @bo911 and I was finally able to build the image without errors, I close this issue.
Feel free to reopen, @bo911

closed 2020-07-02 by heuer