ftCommunity/ftcommunity-TXT #171: TXT network bug

Kry9t0n, 2019-07-29:
Currently I´m using Firmware Version 0.9.5 and several times after booting the TXT can´t login into the local WLAN although before he was online and the local network is shown in the info tab. 
It´s possible get access to the network again by going into the network app and setting the ip-address to static and setting it back to automatic. Then suddenly the status in the info tab changes from WLAN: on, no IP-Address to WLAN: on, [IP-Address]. Also in later Versions of the Firmware this bug occured.

Kry9t0n, 2019-08-14:
I´m currently trying to get more information about the behaviour of the dhcp client in the situation of a crash. I´ve tried "cat /var/log/syslog | grep -Ei 'dhcp' " but there is not syslog on the TXT. Any advice?

hypnotoad, 2022-06-12:
Can you try a 0.10 release? It should have a re-worked dhcp and wifi connect logic.

ski7777, 2022-12-19:
closing due to no activity and no reproduceability

closed 2022-12-19 by ski7777