ftCommunity/ftcommunity-TXT #173: openCV bug in cfw

PeterDHabermehl, 2019-10-14:
Hi together,

My attention was brought to this:

https://stackoverflow.com/questions/31996367/opencv-resize-fails-on-large-image-with-error-215-ssize-area-0-in-funct#32032784

Which prevents some apps (Cube solver, TXTShow) from working correctly.

Obviously the workaround mentioned in the link above would have to be implemented in the cfw.

Kind regards
Peter

harbaum, 2020-04-21:
Naja ....

pip3 install opencv-contrib-python
...
/opt/ftc/apps/user/cb8adf72-f23b-11e6-bc64-92361f002671/benoitxt.py
ImportError: libcblas.so.3: cannot open shared object file: No such file or directory
...
sudo apt-get install libatlas3-base
...
/opt/ftc/apps/user/cb8adf72-f23b-11e6-bc64-92361f002671/benoitxt.py
ImportError: libjasper.so.1: cannot open shared object file: No such file or directory
....
sudo apt-get install libjasper1
....
/opt/ftc/apps/user/cb8adf72-f23b-11e6-bc64-92361f002671/benoitxt.py
Speicherzugriffsfehler

:-(

harbaum, 2020-05-06:
May I suggest to make one of the old and working versions the 'official' one as long as this problem is not solved and delete the broken newer versions?

I am regularly facing disappointed users and had to remove the cube app from the store to prevent further confusion. 

This situation now last for quite some time and it is imho pretty annoying given the fact that this prevents a whole bunch of apps from working

hypnotoad, 2022-04-04:
Hi Peter. An OpenCV update would be nice, of course, but I have the impression that this particular problem could maybe be solved in an easier way (an image area of 2^31=2B pixels would not fit into the TXT memory). Do you have some short code to trigger the bug?

PeterDHabermehl, 2022-04-04:
I can't even remember what exactly this was about. I think there was some issue running the cube solver app on recent community firmware releases due to some OpenCV issues, being tracked down to the image scaling functions.
Maybe @harbaum remembers better?

harbaum, 2022-04-05:
I also have only vague memories. Imho none/few of the apps using opencv were working. This included the cube solver but imho also other apps like the barcode scanner or the camera viewer. Imho also Peters image Viewer was affected

ski7777, 2022-12-19:
due to closing #180 I would close this too

closed 2022-12-19 by ski7777