ftCommunity/ftcommunity-TXT #176: Replace libroboint / robointerface.py by original code

PeterDHabermehl, 2020-04-30:
Erik, the author of libroboint, has migrated his software to gitlab:
https://gitlab.com/Humpelstilzchen/libroboint

Furthermore, the python module was updated for Python 2 and 3 compatibility. The patched module of the cfw is obsolete.

So the cfw Python module should be removed and libroboint and its Python module should be sourced direcly from the gitlab account mentioned above.

heuer, 2020-04-30:
Side note: A little trick is still necessary because cmake is only looking for Python 2.

See: <https://github.com/ftCommunity/tx-pi/commit/513c95d8b6d1a1faf634e60bc7c7d46be3dbb5b7#diff-07c71bdb53d0b9ef903aa8a18cab5c3eR530>

libroboint could possibly add a target "make install python3".

ski7777, 2022-12-09:
Can we fix this locally with a patch and try to merge it there later?

ski7777, 2022-12-30:
Postponing this to 1.1

ski7777, 2022-12-30:
First try to bundle libroboint/python-libroboint is uploaded. Sadly python installs to host dir. Should be easy to fix but not tonight

ski7777, 2023-06-29:
main topic of this issue fixed in 1f3adefe96d1c7ec1bf4064593d999f9e1b75b40

closed 2023-06-29 by ski7777