ftCommunity/ftcommunity-TXT #177: Call for participation: Bump kernel to 5.4.x

ski7777, 2020-05-01:
Since a few weeks I´m trying new stuff with the CFW and one major point is to bump the CFW to Kernel version 5.4.x
Sadly there were some breaking changes:

- NAND/MTD is not accessible anymore
- Display unusable

I am not able to find the problems...
I hope someone is able to debug and solve it.

For testing just set:
`BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="v5.4.36"`

Raphael

heuer, 2020-05-02:
What is the purpose of this exercise? Does the kernel offer any advantages on the device?
If this version creates more problems than solving, why not keep 4.x?

ski7777, 2020-05-02:
According to https://www.kernel.org/category/releases.html we could stay on 4.14 until 01/2024

PeterDHabermehl, 2020-05-02:
Would the kernel bump fix openCV? Or would even more apps stop working?

ski7777, 2020-05-02:
This is about the kernel update not openCV. You might compare it with updating from Windows 10 2016 to Windows 10 2020

PeterDHabermehl, 2020-05-02:
It was more of a rhetorical question. I would prefer someone fixing existing issues instead or raising new ones.I hate to see my apps not working because of fw issues and we all know that others left the team for the same reason.

ski7777, 2020-05-02:
I don't want to raise new issues. I never used openCV because I don't have a camera...

PeterDHabermehl, 2020-05-02:
But that's not a justification for ignoring the issue. We had several complaints from cube solver users who reported the app not working and the only answer for more than a year now is the advice to downgrade to a 0.93.something release. -- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

Domi04151309, 2020-05-02:
> It was more of a rhetorical question. I would prefer someone fixing existing issues instead or raising new ones.I hate to see my apps not working because of fw issues and we all know that others left the team for the same reason.

Updating the codebase is still important because it reduces the technical dept and can prevent a lot of additional work that would have to be done otherwise.

harbaum, 2020-05-06:
Updating the kernel or anything else in the cfw is totally useless if you constantly break things that way and render many applications useless by those 'updates'.

harbaum, 2020-05-06:
This kind of updates without real understanding and without proper testing causes things like issue #174

ski7777, 2020-07-27:
Help is on this is still very appreciated :)

harbaum, 2020-07-28:
Have the current problems been fixed in the meantime? Unless the problems introduced by previous updates have been fixed further updates will likely cause additional problems and should imho be avoided for now.

ski7777, 2020-07-28:
Have you fixed anything or even contributed in any way?

ski7777, 2022-12-19:
closing this due to ongoing participation in new-buildroot branch

closed 2022-12-19 by ski7777