ftCommunity/ftcommunity-TXT #178: Non-issue: Buildroot solution for TX-Pi

heuer, 2020-05-05:
Regarding the todo in the roadmap <https://github.com/ftCommunity/ftcommunity-TXT/projects/1#card-15597781>

IMO we're currently quite happy with the different solution. 

You shouldn't put too much effort into this proposal.

heuer, 2020-06-10:
I think it is counterproductive to continue working on the buildroot solution. 

The TX-Pi project is satisfied with the current solution and would lose a lot of flexibility by switching to buildroot. 

We will not give up our solution in favor of buildroot and I think it would be counterproductive if the cfw project would offer its own TX-Pi images that are fundamentally different from our Debian solution.

PeterDHabermehl, 2020-06-10:
I will not support a buildroot setup for TX-Pi.

There are too many issues with the TXT community firmware that don't get fixed
and keep me already from supporting the TXT cfw.
I don't want the same to happen to the TX-Pi. I can handle the setup-script,
but I won't work myself into buildroot just to fix what others have broken.

Am Mittwoch, 10. Juni 2020, 09:09:01 CEST schrieb Lars:
> I think it is counterproductive to continue working on the buildroot
> solution.
>
> The TX-Pi project is satisfied with the current solution and would lose a
> lot of flexibility by switching to [...]

heuer, 2020-06-10:
@ski7777, just to make it clear: All members of the TX-Pi project are against the buildroot solution. 

In my opinion it would be unfair to the TX-Pi project to create another TX-Pi distribution based on buildroot, because it does not correspond to the intention of the TX-Pi project.

ski7777, 2020-06-10:
We could still have a Buildroot-based solution and for advanced users a Raspberry Pi OS based solution.
Currently the tx-pi project is splitting apart from the CFW. There are special features being introduced on TX-Pi using apt without trying to implement the same on the CFW, there is a seperate app store with "TX-Pi-only"-apps but most of them would run on CFW as well.
I think we should reunite these projects to regain intercompatibility. It might even be possible to create sd-card images that can be put into TXT and Pi.
If there are any components that are downloaded via apt we should thinking about either building it into the CFW directly or provide packages for that.

Please let me now if there are any point I forgot about

Regards
Raphael

ski7777, 2020-06-10:
> I will not support a buildroot setup for TX-Pi. There are too many issues with the TXT community firmware that don't get fixed and keep me already from supporting the TXT cfw. I don't want the same [...]

You are absolutely right about that and I think we should fix those problems in the first place. I working very hard on fixing those problems an I´m sure most problems will be fixed soon.
Don't hesitate to open an issue even if it's a minor problem

heuer, 2020-06-10:
In my opinion it is not a good idea to create two different TX-Pi distributions. It only causes confusion among potential users:
* Which distribution should I use? 
* What are "advanced" users? 
* Why does X work under distribution A but not under distribution B?
* ...

We can simply avoid these questions and confusion by burying the buildroot idea since it does not solve any problem but may cause more problems.

I also don't see any advantages over the current approach. 
The TX-Pi project is very careful with the TX-Pi specific apps. All apps that run under both devices are only published in the cfw store.The TX-Pi store only contains apps that require the TX-Pi or that demonstrate the power of the Raspberry Pi hardware.

I think it would be wiser to put the energy into the cfw and not create artificial problems that do not exist yet.

harbaum, 2020-06-30:
I also fully agree. By this kind of 'technical improvements' the CFW has gotten into a pretty unusable state for most users.  This two things are IMHO important: a) time spent working on the CFW ist best spent on fixing it and getting it into the usable state it was in before the recent 'iptomizations'. And b) we need to make sure that the TX-Pi setup doesn't face the same fate.

Since I don't plan to use the CFW on the TXT again my main focus lies on b).

Using separate app stores will get more and more important since I don't plan to keep my app compatible with the CFW. So they will likely become unusable on the TXT (most of them already are since the CFW is broken atm) and will only run on the TX-Pi.

I will definitely try to prevent

hypnotoad, 2022-04-04:
I am not familiar with the TX solution. Would it be possible to discard buildroot for the CFW/TXT and use a Debian-based solution there as well (it is armv7a, or?)?

PeterDHabermehl, 2022-04-04:
It is not fischertechnik TX, but the Raspberry Pi based TX-Pi project. TX-Pi brings (kind of) community firmware compatibility to a touchscreen-equipped Raspberry Pi, see [http](https://tx-pi.de/de/)
TX-Pi setup is basically installing a certain Raspbian light release plus running a shell setup script. Unfortunately also TX-Pi is suffering from a lack of participation recently...
Anyway this discussion was about creating a buildroot project for TX-Pi instead of maintaining the setup script, which was declined by the people taking care of TX-Pi at that time. I guess this issue may be closed. TX-Pi is two or three Raspbian releases behind and it's unclear if ever someone, including me, will ever try to update it. Which is kind of sad.

ski7777, 2022-12-09:
> TX-Pi is two or three Raspbian releases behind and it's unclear if ever someone, including me, will ever try to update it. Which is kind of sad.

I think it is the other way around. It would be so nice to make this project run on other platforms as well. After migrating to one platform, other platforms should not be a big deal.
Has anyone ever tried running a custom firmware on the TXT 4.0?